reyntjensw/Homey-FusionSolar #1: Error

SR253, 2022-03-17:
After login I get "Cannot read property 'name' of undefined" when looking for devices.

reyntjensw, 2022-03-17:
@SR253 when adding a new device, the first screen is a login screen where you need to enter your credentials.
Once that is done, the devices will appear. It is important to hit the green login button.

SR253, 2022-03-17:
> @SR253 when adding a new device, the first screen is a login screen where you need to enter your credentials. Once that is done, the devices will appear. It is important to hit the green login button.

I am logging in, next page it states the "Cannot read property 'name' of undefined" immediately.

reyntjensw, 2022-03-17:
@SR253 can you also send a debug log or provide me with your credentials via email? ***@***.***)

reyntjensw, 2022-03-18:
@SR253 I did see the diagnostic report coming in and I assume you have the same issue as Jorgen does (https://github.com/reyntjensw/Homey-FusionSolar/issues/2)

Can you provide me with your password please via email?

There is a new version of the app that is under approval which will give you a better error reporting.

SR253, 2022-03-18:
I'm in contact with Huawei to ensure that the API is activated. I will get back to you when I got it confirmed.

SR253, 2022-03-18:
> @SR253 I did see the diagnostic report coming in and I assume you have the same issue as Jorgen does (#2)
> 
> Can you provide me with your password please via email?
> 
> There is a new version of the app that is under approval which will give you a better error reporting.

I'm in contact with Huawei to ensure that the API is activated for me. I will get back to you when I got it confirmed.

reyntjensw, 2022-03-22:
@SR253 please let me know if you need extra assistance

SR253, 2022-03-29:
HI, I have API access now and can access my station via Postman. 

The Homey app though gives me an error…

 

 

 

/Stefan

Reply to: reyntjensw/Homey-FusionSolar ***@***.***>
Date: Tuesday, 22 March 2022 at 06:34
To: reyntjensw/Homey-FusionSolar ***@***.***>
Cc: S R ***@***.***>, Mention ***@***.***>

reyntjensw, 2022-03-29:
Hi Stefan

What is the error?

SR253, 2022-03-29:
The Homey icon shows a warning sign and shows this error...
[image]

SR253, 2022-03-29:
I contacted Huawei support, maybe they fixed something because now it seems to be working fine...

SR253, 2022-10-11:
My server address is: eu5.fusionsolar.huawei.com

 

Not intl.fusion….

 

/SR

Date: Tuesday, 29 March 2022 at 18:30

reyntjensw, 2022-10-11:
@SR253 did you get anything back from Huawei support?
They explicitly requested to use the intl instead of a specific server.

Can you please mail me your ticket number from Huawei or can you give me some more information of what came out of the ticket?

SR253, 2022-10-11:
I have not followed up on it. Seems random which servers to use. Sometimes one work, then mext day another. Today the server: region02eu5 works…I dont use it in any flow at the moment so not priority for me to get fixed./SRSkickat från min iPhone11 okt.

reyntjensw, 2022-10-24:
@SR253 so you were able to add it into the homey?

SR253, 2022-10-24:
Yes, but I needed approwal from Huawei to use the api. Took some effort to get that. Needed the installers approval in writing to send to Huawei and not anyone could write it needed to be the company owner. /SRSkickat från min iPhone24 okt.

closed 2022-10-24 by reyntjensw