reyntjensw/Homey-FusionSolar #3: Control charging based on prices

mackewigren, 2022-03-25:
Hi
I am about to invest in probably a Huawei inverter. In Homey I know when prices are high and low from my electrical company. Is it possible to control the inverter that when the price is low it should charge instead of having schedules in the inverter? More or less every hour tell the inverter if it is a charging hour or a discharging hour?

reyntjensw, 2022-03-29:
@mackewigren from what I know this is not possible.
The inverter in my case is only charging with solar power.

You can create a case at Huawei and ask them if the APi has this ability, if so it can be added.

mackewigren, 2022-03-29:
> @mackewigren from what I know this is not possible. The inverter in my case is only charging with solar power.
> 
> You can create a case at Huawei and ask them if the APi has this ability, if so it can be added.

According to documents I have seen it should be possible to charge with schedules. But I guess maybe not controlled via API.

reyntjensw, 2022-05-19:
@mackewigren after discussing this with Huawei, there is no way in doing this.

reyntjensw, 2022-10-11:
Good morning

From what I know, charging cannot be controlled with the api.
The system makes "intelligent" decisions based on the load.

Wouter

mackewigren, 2022-10-11:
> Good morning From what I know, charging cannot be controlled with the api. The system makes "intelligent" decisions based on the load. Wouter
> […](#)

Thanks. There are different settings. I have TOU as Operation mode. Then I can define several time slots during the day to control if I should Charge or Discharge. See attached image of example settings. It would be good enough to just have two time slots a day. One for the coming hour and one for the rest. And every hour it should be updated based on a workflow.

It should be possible with MODBUS TCP from what I heard. But I don't know anything about that. 

[image]

mackewigren, 2022-10-26:
Hi

I have had a conversation with Huawei and they didn’t know themselves. But when I asked if they had any plan for adding battery charging control in the API the actually found that version 7 DOES SUPPORT charging. I am not good enough at API but think I understood it as chapter "6.3.6 Interface for Delivering Battery Charge and Discharge Tasks” is about this.

Would be nice if possible to based on any trigger in Home just send ”Start charging” or ”Stop charging”. Or in combination with ”Start charging for x minutes”

What do you think about this? Must be one of the most wanted functions.

Best regards

Marcus

> 
> 
> @mackewigren <https://github.com/mackewigren> Thanks for the info.
> Currently, I'm not sure if I will incorporate ModBus because it seems like they are going to discontinue 3rd party integrations. (https://forum.huawei.com/enterprise/en/forum.php?mod=redirect&goto=findpost&ptid=789585&pid=4632813)
> 
> I'll put this one on the backlog
￼

reyntjensw, 2022-11-10:
Hi @mackewigren 

It will not be impossible if you are referring to point 6.3.6.
They have mixed 2 versions of the api in the documentation.

If the request url includes a port like this one, it is not possible.
[image]

mackewigren, 2022-11-11:
Hi

I am not sure exactly what you mean. You write both ”not impossible” and also ”it is not possible”. I think you mean that it is NOT possible.

Why is it not possible based on a port in the URL? Most of the request urls in the document have port number

Best regards

Marcus

reyntjensw, 2022-11-11:
Hi @mackewigren 

If you look at the code (https://github.com/reyntjensw/Homey-FusionSolar/blob/main/drivers/luna/api.js#L6) this is the url I am using to do all the calls, it starts with thirdData. This api is publicly available, the one in the document is not reachable, I think it's created for another platform.

mackewigren, 2022-12-13:
Hi

I have had the ability to test this now from Postman. And it works. I could trigger battery charging. I use the same credentials but Huawei needed to ”open” access for this version. 

So it definitely should be possible. Would you be interested in developing for battery control?

Best regards
Marcus

-------------------------------------------------------------
Marcus Wigren	0733-983138
Rosenvägen 5	        
23735 Bjärred	        ***@***.***

reyntjensw, 2022-12-19:
@mackewigren I right now don't have the time to program this. If someone else has the time, please create a PR.

trapeze78, 2023-01-01:
Hi, I have a Homey, recently bought a LUNA inverter and will soon get som Huawei batterier as well. I share the exact same situation as @mackewigren and will follow this thread closely. Not sure if I am able to program this myself but will dig into the API details as soon as everything is up and running. Option would be to run something using a Home Assistant setup or wait for my electricity company Tibber to get it done themselves but that might take a while. Keep up the good work!

boboiusa, 2023-03-27:
> Hi I have had the ability to test this now from Postman. And it works. I could trigger battery charging. I use the same credentials but Huawei needed to ”open” access for this version. So it [...]

@mackewigren 
Hello Marcus.
I will soon get my Huawei battery and is was wondering if you have had any more luck regarding smart charging the battery. I am also from Sweden and it would be great to know what is possible with Huawei batteries today. 

Cheers, 
Johan

mackewigren, 2023-03-28:
> 

HI boboiusa

I have not tested it more. I dont have enough knowledge on my own I think. But I tested a luttle from Postman just to make sure it is possible through API. But unfortunately I ave not been able to convince the app author to implement this and I have not enough knowledge I think to do it on my own. 

I think the only thing the app (or a script) needs to do is

1) Change charge mode. Three different options. Charge, Discharge, Nothing. This can then be triggered through flows

2) Could be nice to set electricity price so it would be possible to see revenue in the Fusion Solar app. 

The other settings is probably overdoing it. Feed-to-grid or Feed-to-battery can be changed in the Fusion Solar app is my opinion.

So if you want to try to convince the author, just go on :-)

mackewigren, 2023-03-28:
> If someone can give me the commands, I can try to implement them.

Hi, I will get back about this in a couple of days. Unless someone else have the time before me. Thanks a lot.

boboiusa, 2023-04-01:
Thank you all for trying to get this to work.

Have a nice weekend!

mackewigren, 2023-04-04:
Hi
WIll this be "good enough" as examples? See attached pdf.
[Fusion Solar Battery Flow.pdf](https://github.com/reyntjensw/Homey-FusionSolar/files/11147423/Fusion.Solar.Battery.Flow.pdf)

reyntjensw, 2023-04-09:
@mackewigren thanks for the example, I'm first working on a login issue, when that is solved I'll try to work on this.

mackewigren, 2023-04-10:
> @mackewigren thanks for the example, I'm first working on a login issue, when that is solved I'll try to work on this.

Just let me know if I can help.

mackewigren, 2023-09-26:
@reyntjensw Hi. Wanted to check if you have had any progress with this? We are many that have “requested” this feature. In different threads.

reyntjensw, 2023-09-26:
@mackewigren for now this is not on the backlog since the app is not stable.
It does not make sense to develop this if the app is not working as expected.