reyntjensw/Homey-FusionSolar #5: Missing import/export to grid

JorgenSeemann, 2022-03-29:
It would be nice to also have the "export/import to grid", so one could activate flow on export (f. ex. turn on an electric water heater or equal)

Cheers :)

reyntjensw, 2022-04-14:
@JorgenSeemann I'll look into it

andiwirz, 2022-05-19:
@reyntjensw I think you should be able to get additional data from "Power Sensor":

1. step
https://eu5.fusionsolar.huawei.com/thirdData/getDevList
{
    "stationCodes": "XYZXYZ",
    "collectTime": 1652997079402
}

use data from response:
"devName": "Power Sensor",
"devTypeId": 47,
"id": WHATEVERITIS

2. step
https://eu5.fusionsolar.huawei.com/thirdData/getDevRealKpi
{
    "devIds": "WHATEVERITIS",
    "devTypeId": "47"
}

3. response:
                "dataItemMap": {
                "active_cap": 1719.28,
                "meter_status": 1.0,
                "meter_i": -1.49,
                "reverse_active_cap": 554.84,
                "power_factor": -0.901,
                "reactive_power": 639.0,
                **"active_power": -2615.0,**
                "run_state": 1,
                "meter_u": 234.3,
                "grid_frequency": 50.02
            }

reyntjensw, 2022-05-20:
Hi @andiwirz 

Thanks for your response, I've implemented this on test, please check it out : https://homey.app/a/com.huawei.solar/test/
Keep in mind that you might need to follow this guide as well https://community.homey.app/t/huawei-solar/61745/2

The Power Sensor was not available on my setup, but a meter was available.

Keep me posted

andiwirz, 2022-05-20:
Seems not to work 😟

[image]

could you update the github source?
I would like to play around with the code if you agree?

reyntjensw, 2022-05-20:
@andiwirz did you check the post on the homey site?
That post fixes the issue you see above.

You can for sure work in the code, the current code can be found in this branch https://github.com/reyntjensw/Homey-FusionSolar/tree/feature/multiple_improvements

Can you please create a new feature based on this one?

Thanks

andiwirz, 2022-05-20:
When I revert to version "1.0.18" it works instantly

reyntjensw, 2022-05-20:
That is normal because  that is only using 1 api. To avoid hitting any quotas it is needed to use 2.

Did you use the article to set your correct server on the test version?

reyntjensw, 2022-05-20:
Ok, what is your current inverter?
Is it a 3phase one or a 1 phase one?

[image]

andiwirz, 2022-05-20:
3phase

andiwirz, 2022-05-20:
[image]

andiwirz, 2022-05-20:
[image]

reyntjensw, 2022-05-20:
Can you please share your credentials?
You can mail them to ***@***.***

On my system, I don't have a power meter, only a meter https://github.com/reyntjensw/Homey-FusionSolar/blob/d49a59d084cd37d318e474349997a26ed4f50cfc/drivers/luna/api.js#L121

I'll push a small change in a couple of minutes

[image]<https://user-images.githubusercontent.com/52567683/169567255-daa1d2af-0f02-4e5f-9863-8e1ca730f763.png>

reyntjensw, 2022-05-20:
Please try out the new version, if it doesn't work I might need your credentials.

https://homey.app/a/com.huawei.solar/test/

andiwirz, 2022-05-20:
[image]

great!!!!

closed 2022-05-23 by reyntjensw