reyntjensw/Homey-FusionSolar #7: log in

Erik2a, 2022-04-12:
i am new to homey.
Trying to login the huawei solar app. Doesn't work. Says credentials are unknown.
Any idea how to make it work?
Thx!!

reyntjensw, 2022-04-14:
Hi @Erik2a 

Good chance that the credentials that you received are incorrect.
There have been 2 cases already where the credentials just don't work.

For example https://github.com/reyntjensw/Homey-FusionSolar/issues/1

Best thing to do is to contact Huawei support and ask them to check it.
You can mail them on this address eu_inverter_support@huawei.com

Erik2a, 2022-04-14:
thank you very much!
Though it should be correct as i use it to login on their app and there is no problem there.
But I'll try to contact them.

reyntjensw, 2022-04-18:
Hi @Erik2a , the login for the app and the api are different logins.

Erik2a, 2022-04-18:
Ja, blijkbaar.  Ik kreeg net dat antwoord van Huawei.
Ze hebben het over een api account .
Maar dat heb ik niet. Ik ben  helemaal nieuw hierin.
Ik weet wel wat een api is Maar heb er zelf nooit één ontwikkeld.
Groetjes
Erik

reyntjensw, 2022-04-18:
Dag Erik

Kan je me laten weten wat Huawei support doorgaf?

Erik2a, 2022-04-18:
https://eu5.fusionsolar.huawei.com
 
Do you need and openAPI account? You need to have an installer account that has access to a pv plant so the openAPI account to be created.

reyntjensw, 2022-04-18:
Kan je inloggen op de url die ze doorgegeven hebben?
Als dit kan zit je al goed.

Je moet bij Huawei een openApi account aanvragen, dit heb je nodig om verder te kunnen.

Erik2a, 2022-04-18:
Sorry, via ander mail adres gestuurd.
Dus dit stuurde ze me terug:

«  
Do you need and openAPI account? You need to have an installer account that has access to a pv plant so the openAPI account to be created. »

Wat is een installer account?
Heb ik dat nodig of moet ik een api account openen? Vraag ik dat aan bij hun?

Groetjes,
Erik

reyntjensw, 2022-04-18:
Het installer account heb je enkel nodig om je pv plant toe te voegen aan hun cloud omgeving.
Als dit al gedaan is, mag je dit negeren.

Erik2a, 2022-04-27:
Hoi Wouter,

Heb eindelijk de nieuwe credentials, had heel wat voeten in de aarde!

Log in lukt maar ik kom er niet in. Ik krijg direct een fout melding:
Device is unavailable
Error retrieving data (typeError:cannot read property ‘0’ of undifined.

Enig idee wat dit betekent?

Groetjes,

Erik

reyntjensw, 2022-04-27:
Dag Erik

Er is een nieuwe build doorgestuurd, je kan de test versie al even installeren en testen https://homey.app/a/com.huawei.solar/test/

Erik2a, 2022-04-27:
Heb net uitgeprobeerd.
Maar nu vind hij helemaal niets:
Select the device you want to add
Cannot read property ’name’ of undefined

Zegt je dat iets?

Erik2a, 2022-04-27:
Nog een vraagje (als je het vervelend vind, zeg het dan!!)
Ik werk met de app op mijn smartphone. Dat lukt allemaal.
Wilde nu eens kijken of ook op MacBook lukt, via homey.app.
Die geeft aan dat ik ingelogd ben, ziet dat ik een Homey Pro gekocht heb maar 
zegt tegelijkertijd: cannot find device. En ik kan daar dus niks mee.

Any idea? Of werkt het niet via de computer?

reyntjensw, 2022-04-27:
Dag @Erik2a , kan je me je credentials even doormailen aub? 
Dit mag naar wouter@chismoor.net

reyntjensw, 2022-04-27:
Probleem met @Erik2a overlopen en opgelost.

closed 2022-04-27 by reyntjensw