reyntjensw/Homey-FusionSolar #9: Missing Discharge battery power

Ktlhouben, 2022-04-30:
[image]

reyntjensw, 2022-05-02:
@Ktlhouben to enable the battery to have to go into the settings of the app, settings icon at the top and enable the battery there. Then wait about 5 minutes and it will pop up. The current yield should pop up if there is sunshine

reyntjensw, 2022-05-02:
@Ktlhouben your inline image is not showing, can you please post that?

Ktlhouben, 2022-05-02:
[image]

reyntjensw, 2022-05-03:
@Ktlhouben I have released a new version, please install this.
If the issue still exists, please send me diagnose info, you can find this via more (...) -> apps -> Huawei Solar -> wheel icon -> generate report

Ktlhouben, 2022-05-03:
Dear Wouter,

After the Update Homey can't make connection at all.
As you asked for the generated code below.

ececf538-cc5f-4a26-85f7-8e1840d9fb66

[image]

reyntjensw, 2022-05-03:
Great, could you tell me what kind of inverter you have? Do you also have a battery?

[image]<https://user-images.githubusercontent.com/47369021/166486647-4711d7f2-89d6-488e-b02c-44d1f3c7e57d.png>

Ktlhouben, 2022-05-03:
Hello Wouter.

batterij
merk HUAWEI
type LUNA 2000 10-S0
bruikbare capaciteit 10kW

Omvormer: HUAWEI SUN2000-4,6KTL-L1, max. AC vermogen: 4600 VA
Totaal max AC vermogen: 4600 VA

reyntjensw, 2022-05-03:
Can you please try to remove the app and reinstall it?
If that does not work, can you please provide me with your credentials please?

You can send them to wouter@chismoor.net

reyntjensw, 2022-05-09:
The problem was solved and tested by @Ktlhouben

Ktlhouben, 2022-05-30:
Hoi Wouter,

Sinds de laatste update ligt de app bij mij weer offline.

Hetzelfde probleem als voorheen.
Reeds verwijderd en opnieuw geïnstalleerd maar zonder resultaat.

reyntjensw, 2022-05-30:
hi @Ktlhouben 

Is het mogelijk om na te kijken of de server correct staat?
 https://community.homey.app/t/huawei-solar/61745/2

Ktlhouben, 2022-05-31:
Goedemorgen Wouter,
Deze staat juist.

Met vriendelijke groeten,

Kristof Houben

https://community.homey.app/t/huawei-solar/61745/2

reyntjensw, 2022-06-12:
Kristof

Er is een nieuwe versie van de app online gezet, kan je deze even testen aub?

Ktlhouben, 2022-06-13:
Hoi Wouter,
Ook deze doet het niet.

f3814f75-ac85-48c3-87b2-85a7ba68529a

reyntjensw, 2022-06-19:
@Ktlhouben ik zie dit in de logs terugkomen :
`getDevRealKpi
{ failCode: 305, immediately: true, message: 'USER_MUST_RELOGIN' }`

Kan je even contact opnemen met Huawei support en vragen waarom je deze foutmelding krijgt?
De functie die aangeroepen wordt is **getDevRealKpi**

reyntjensw, 2022-12-20:
@Ktlhouben is this issue still present?
Can you please install the latest version and try again?

Ktlhouben, 2022-12-20:
Hello Wouter,

The issue is still present.

Ktlhouben, 2022-12-20:
d8271c89-6bac-4c68-a9cd-9f6d06b1cba0

reyntjensw, 2023-01-22:
@Ktlhouben did you get in touch with Huawei support?

Ktlhouben, 2023-01-22:
No why? should I?

Met vriendelijke groeten,

Kristof Houben

reyntjensw, 2023-01-23:
Hi @Ktlhouben you should because the api is not returning the correct data.

Ktlhouben, 2023-03-11:
It works fine right now.
Thanks Wouter for your help.

closed 2023-03-12 by reyntjensw