reyntjensw/Homey-FusionSolar #10: Cannot read property error after upgrade

JorgenSeemann, 2022-05-17:
Diag log - ca100552-a4b6-411e-afb1-d1823a1f1aab
Tried reboot app and homey, but this didn't help:/

reyntjensw, 2022-05-17:
@JorgenSeemann I am seeing this in the logs :

```
Huawei has been initialized
{"data":null,"success":false,"failCode":20001,"params":{"currentTime":1652765355424},"message":null}

```
Meaning there is something wrong with your login on the new api endpoint.

Please install https://homey.app/a/com.huawei.solar/test/ which is using the old api.

I have updated the case to Huawei and as soon as I have info, I'll let you know

reyntjensw, 2022-05-19:
hi @JorgenSeemann 

Can you login into the web interface of Fusion solar and tell me what the url is when logging in?

When I login I use https://eu5.fusionsolar.huawei.com/, after login my url changes to https://region01eu5.fusionsolar.huawei.com/.

Can you tell me what your region0x server is?

reyntjensw, 2022-05-19:
@JorgenSeemann please install this new version https://homey.app/a/com.huawei.solar/test/

And also follow this guide https://community.homey.app/t/huawei-solar/61745/2, this should solve the issue

reyntjensw, 2022-05-20:
@JorgenSeemann can you confirm that the issue is solved please?

JorgenSeemann, 2022-05-20:
Yes, it seems to be working as expected 😀

reyntjensw, 2022-05-20:
Great thanks!

closed 2022-05-20 by reyntjensw

JorgenSeemann, 2022-10-11:
Hi Wouter

Having logged in, my URL shows server region02eu5

Br
Jørgen

reyntjensw, 2022-10-11:
Hi Jorgen

You can change it by going to the settings in the app, there you have the option to change the server.

Wouter

reyntjensw, 2022-10-11:
[image]