reyntjensw/Homey-FusionSolar #12: Cannot read property 'name' of undefined

Willemss001, 2022-05-20:
Dear,

Hope you’re well and able to help.

I'm trying to add my device but getting below error after login with my newly created API account;
Cannot read property 'name' of undefined
Any thoughts?

Thanks in advance!

Kind regards,
Stefan

reyntjensw, 2022-05-20:
@Willemss001 

This issue might be related to your account that is not created correctly by Huawei.
Please contact them again and ask them to verify if everything is ok.

This should fix it.

Wouter

Willemss001, 2022-05-20:
Hi Wouter,

Thanks for getting back to me that quick.

I believe the account is not the issue.
Following Huawei’s doc I was able to query my inverter using Sandman app.
***@***.***

Any thoughts?

Thanks,
Stefan

reyntjensw, 2022-05-21:
@Willemss001 can you show me a screenshot of the login process please? 
The body is what I'd like to see

[image]

Willemss001, 2022-05-23:
Not sure what you did to have this fixed but you did! 😉
Thanks to  lot!

Willemss001, 2022-10-11:
Here you go .. anything else you want me to check or provide you?

***@***.***

Thanks,

[image]<https://user-images.githubusercontent.com/4648544/169633960-28396f26-bf42-495f-a094-688dc7003474.png>

reyntjensw, 2022-10-24:
@Willemss001 I don't see any output, could you post  it here?

reyntjensw, 2022-12-20:
@Willemss001 is this issue still present? Can you please install the latest version and try again?

reyntjensw, 2023-01-22:
@Willemss001  can I close this issue?
Was your problem solved?

Willemss001, 2023-01-24:
Dear Wouter,

Hope you're well and thanks for your support.

Yes, you may close the ticket. The actual problem got resolved already months ago by Huawei's support. I believe I did confirm the ticket closure via the community, apparently not so apologies for that.

Kind regards,
Stefan

closed 2023-01-24 by reyntjensw