reyntjensw/Homey-FusionSolar #14: Cannot read property 'name' of undefined

Samueljohan, 2022-07-07:
Hi,
I can not login with this app,
With fusion solar is no problem. 
What iam i missing?

//Samuel

reyntjensw, 2022-07-08:
hi @Samueljohan 

Could you please send me the diagnostics report in the app?
You can find it under apps -> fusion solar -> diagnostics report

Samueljohan, 2022-07-08:
Hi,

6ec9cbff-737f-4bf0-8691-33c8623f6d96

Best Regards
Samuel Johansson
Falkenbergs Elektriska AB
Kvekatorpsvägen 20
31132 Falkenberg
Sweden

reyntjensw, 2022-07-08:
Good morning

I see this  in the logs 

`getSystems
{
  errorCode: '49401021002',
  exceptionInfo: 'Request is rejected by api quotaControl. api:ies_PVMSNbiService_thirdData_1.0.74, key:/getStationList_POST'
}
`

The  request is being throttled by the Huawei api, can you wait and try in about 30 minutes?

Samueljohan, 2022-07-08:
Hi,

 

Okey.

I tried now but still the same.

 

 

Best regards

 

Samuel Johansson

Marine Electrician

 

 

FALKENBERGS ELEKTRISKA AB

Kvekatorpsvägen 20

311 32 Falkenberg

 

+46 (0) 709 63 29 42

+46 (0) 346-132 50

 <http://www.fbgel.se/> www.fbgel.se

 

Orgnr: 556152-0858

VAT: SE556152085801

Inovice: ***@***.*** ***@***.***> 

 

Detta e-postmeddelande har sökts igenom efter virus med antivirusprogram från Avast.
https://www.avast.com/antivirus

Samueljohan, 2022-07-12:
Hi,
See new diagnostics report:
a66487-013e-4370-bee0-b43b90a90fb6

reyntjensw, 2022-07-12:
Samuel

Can you please provide me with your password please?
You can share it with wouter@chismoor.net

Wouter

reyntjensw, 2022-07-13:
@Samueljohan you are welcome

Tobias-Georg, 2022-10-02:
Hello, I also get this " errorCode: '49401021002', exceptionInfo: 'Request is rejected by api quotaControl..." but I am using the the API maybe once every 10 min and until now always different urls.
What is the quota? If I only can call the API every 30min, this is a bad restriction!

reyntjensw, 2022-10-03:
Hi Tobias

The API restrictions are unclear because they have 2 API quotas, one for an older and one for newer systems.
And they are not documented. What you can do is create a ticket and request it from Huawei.

I'm looking at another path, but that is not easy to implement.

ludwich, 2022-12-18:
@Tobias-Georg this should be solved in a PR that is up. key:/getStationList_POST is the problem with the quota and that should be mitigated in that PR.

reyntjensw, 2022-12-20:
@Tobias-Georg the new version is live.
Thanks @ludwich for the PR.

reyntjensw, 2023-01-22:
@Tobias-Georg can I close this issue?
Was your problem solved?

Tobias-Georg, 2023-01-23:
I am now using the Fusion Solar API without problems. Thanks for your support! You can close the issue.

closed 2023-01-23 by reyntjensw