reyntjensw/Homey-FusionSolar #18: Device unavailable after ticking on battery in settings

Mimomaj, 2022-11-15:
Hi, could you please take a look at my problem?
After I tick a battery option in the settings of the app, when the data is updated (5 mins or restart of the app) the device becomes unavailable: 
[image]

My setup:
inverter - SUN2000-5KTL-M1
battery - LUNA2000 5kWh

Mimomaj, 2022-11-15:
diagnostic report: e85a666b-9998-4f22-91e7-d778f0038e58

reyntjensw, 2022-11-15:
Hi @Mimomaj 

Can you share your api key via email please? wouter@chismoor.net

I need to investigate

reyntjensw, 2022-12-20:
@Mimomaj can you please install this version and try again? https://homey.app/a/com.huawei.solar/

Mimomaj, 2022-12-20:
@reyntjensw sure, I'll test it tomorrow

ludwich, 2023-01-02:
@Mimomaj is this working? Close this if it is.

Mimomaj, 2023-01-02:
@ludwich sorry I forgot to post update.
Unfortunately no, same behavior as previous version.

reyntjensw, 2023-01-02:
@Mimomaj are you already using the latest version? It was released this morning

Mimomaj, 2023-01-02:
Yes latest version 1.0.39
[image]

reyntjensw, 2023-01-08:
> Hi @Mimomaj
> 
> Can you share your api key via email please? [wouter@chismoor.net](mailto:wouter@chismoor.net)
> 
> I need to investigate

@Mimomaj can you share this data please?

ludwich, 2023-01-08:
And error logs would show more info now.

reyntjensw, 2023-01-08:
The system sees the battery but is unable to query it.
It keeps return a 305 error, for this I think you better contact Huawei support, unless @ludwich has another suggestion?

`[log] 2023-01-08 16:05:33 [ManagerDrivers] [Driver:luna] [Device:a4b280a6-2c12-45b5-890e-149a6216f43f] devListpowerMeterId {
  devName: 'Battery-1',
  devTypeId: 39,
  esnCode: null,
  id: 1000000035658220,
  invType: null,
  latitude: 48.224056,
  longitude: 17.242726,
  optimizerNumber: null,
  softwareVersion: null,
  stationCode: 'NE=xxxxx'
}
apiData getDevRealKpi values:  { failCode: 305, immediately: true, message: 'USER_MUST_RELOGIN' }
[log] 2023-01-08 16:05:33 [ManagerDrivers] [Driver:luna] [Device:a4b280a6-2c12-45b5-890e-149a6216f43f] devRealKpiBattery null
apiData getDevRealKpi values:  { failCode: 305, immediately: true, message: 'USER_MUST_RELOGIN' }
`

ludwich, 2023-01-08:
hmmm user must relogin. That is weird. Will look at retry logic.

ludwich, 2023-01-08:
Does postman work?

reyntjensw, 2023-01-08:
In postman you get an empty list 
`{
    "data": [
        {
            "devId": 1000000035658220,
            "dataItemMap": {}
        }
    ],
    "failCode": 0,
    "message": null,
    "params": {
        "currentTime": 1673195771631,
        "devIds": "1000000035658220",
        "devTypeId": 39
    },
    "success": true
}`

ludwich, 2023-01-08:
Then support have to step in.

Mimomaj, 2023-01-08:
Thanks guys for spending time on this. I'll contact the support. I'll share with them the postman response right?

ludwich, 2023-01-08:
Yes share the response and they should hopefully fix this.

Mimomaj, 2023-01-08:
OK thanks!

reyntjensw, 2023-01-22:
@Mimomaj  can I close this issue?
Was your problem solved?

Mimomaj, 2023-02-06:
Hi @reyntjensw , could you please share with me the request that you did in postman? Huawei support can't follow up on this without that

reyntjensw, 2023-02-06:
@Mimomaj can you create another diagnostics report please?

Mimomaj, 2023-02-07:
@reyntjensw here it is 5d0c1f84-afa0-47a0-b559-55f89df1a486

reyntjensw, 2023-02-07:
@Mimomaj here you go, the error we get is :
`---- stdout ----
[log] 2023-02-07 07:32:42 [HuaweiSolar] HuaweiSolar has been initialized
[log] 2023-02-07 07:32:43 [ManagerDrivers] [Driver:luna] Luna has been initialized
[log] 2023-02-07 07:32:43 [ManagerDrivers] [Driver:luna] [Device:e76721d3-18c1-48f3-8767-c3cc1aeb905d] Huawei has been initialized
server:  region02eu5
[log] 2023-02-07 07:32:44 [ManagerDrivers] [Driver:luna] [Device:e76721d3-18c1-48f3-8767-c3cc1aeb905d] get basic Data
[log] 2023-02-07 07:32:44 [ManagerDrivers] [Driver:luna] [Device:e76721d3-18c1-48f3-8767-c3cc1aeb905d] get current Data
[log] 2023-02-07 07:32:44 [ManagerDrivers] [Driver:luna] [Device:e76721d3-18c1-48f3-8767-c3cc1aeb905d] devListpowerMeterId {
  devName: 'Battery-1',
  devTypeId: 39,
  esnCode: null,
  id: 1000000035658220,
  invType: null,
  latitude: 48.224056,
  longitude: 17.242726,
  optimizerNumber: null,
  softwareVersion: null,
  stationCode: 'NE=35658216'
}
[log] 2023-02-07 07:32:45 [ManagerDrivers] [Driver:luna] [Device:e76721d3-18c1-48f3-8767-c3cc1aeb905d] devRealKpiBattery {}

---- stderr ----
[err] 2023-02-07 07:32:45 [ManagerDrivers] [Driver:luna] [Device:e76721d3-18c1-48f3-8767-c3cc1aeb905d] Unavailable (Error: measure_battery InvalidTypeError: undefined is undefined but expected number (invalid_type))`

When we look at the code, this is triggered at line https://github.com/reyntjensw/Homey-FusionSolar/blob/main/drivers/luna/device.js#L112

The api call that is being triggered here is https://github.com/reyntjensw/Homey-FusionSolar/blob/1c07ccf2e3f8fbf45778130609586e4d0c7209a1/drivers/luna/api.js#L137

Mimomaj, 2023-02-07:
@reyntjensw So is that ok if I share all of this with the Huawei support?

reyntjensw, 2023-02-07:
sure this is all public information :)

krosso65, 2023-03-04:
Any News on This? I have the same problem

krosso65, 2023-03-04:
fe3d8dd6-4072-456c-b8ab-56a55df97a67

krosso65, 2023-03-04:
My diag

reyntjensw, 2023-05-02:
@krosso65 what is the update from Huawei?

krosso65, 2023-05-05:
I now got the right permissions. But still the device goes "unavailable" , but sometime when I look it works even if I do nothing for a short while?!?!  have you seen the same or is I alone with this?

reyntjensw, 2023-05-09:
@krosso65 can you send me a diag report please?

reyntjensw, 2023-05-10:
I don't see anything in the logs that could indicate a problem.
I have published a new test version, please try it out

krosso65, 2023-05-11:
Ok, how to install test-version?

Skickat från Outlook för iOS<https://aka.ms/o0ukef>

reyntjensw, 2023-05-20:
@krosso65 can you please read this article and try the web method please? https://community.homey.app/t/app-pro-huawei-solar-smart-energy-for-a-better-life/61745/152?u=reyntjensw

krosso65, 2023-05-21:
Hello
When try to add a device with ”webb” I got invalid login, tried both web user and email
When try to add with ”before or after 2023” using api user I got error in screenshot

/regards Per

Skickat från Outlook för iOS<https://aka.ms/o0ukef>

krosso65, 2023-05-21:
After your last update 2.0.5 it seems to work :)
Missing some values, see screenshot

/regards Per

Skickat från Outlook för iOS<https://aka.ms/o0ukef>

reyntjensw, 2023-05-22:
Good morning, the screenshot did not come through.  If you have a battery, for now you need to restart the app, there is a known bug

krosso65, 2023-05-22:
Hello

Thats worked 😊
How often does the values update?
My Daily yield is very slow (now 9h since last update)
Monthly yield is much faster

/Regards Per

Skickades från E-post<https://go.microsoft.com/fwlink/?LinkId=550986> för Windows

reyntjensw, 2023-05-22:
Every 2 minutes an update is triggered

closed 2023-05-28 by reyntjensw