reyntjensw/Homey-FusionSolar #20: Power by the hour integration

ludwich, 2022-12-13:
Adding Power by the hour so i can track the real money gained by cells is not working. It is possible to add but it does not read any values.

ludwich, 2022-12-14:
@reyntjensw i am a javascript/ts dev as well and could support on modbus feature if you want to. Tried to extend main with more capabilities but get a weird error on includes TypeError: Cannot read properties of undefined (reading '0'). Commented out the code if statements where includes is used by still getting it.

ludwich, 2022-12-14:
Yes when i add measure_power i get the error. Will try something else since it might be a problem with setting a value like that when we have capabilities below it. Will branch and make PR.

ludwich, 2022-12-14:
@reyntjensw yes i am getting this issue from main straight away without any changes.

reyntjensw, 2022-12-14:
> @reyntjensw yes i am getting this issue from main straight away without any changes. my bad since i had the app installed as well i was getting throttled. Removed now and will start again later today,

The error handling regarding the throttling needs to be improved.

ludwich, 2022-12-17:
Resolved

closed 2022-12-17 by ludwich