reyntjensw/Homey-FusionSolar #28: Feature Request: 2 values from Power Meter -> Power by the hour app

andiwirz, 2022-12-20:
Hi

It would be great to have two additional values within the app, to use the data in "Power b the hour" app:

[image]

ludwich, 2022-12-21:
@andiwirz I just got my panels the other week and I can´t see this data on my dashboard. Probably since I haven´t made enough energy to sell anything yet so hard to compare with the values i get from the API.

If anyone else know which values these are in the plz tell me and I can implement this.

andiwirz, 2022-12-21:
I'm not sure, but maybe this could help:

[image]

ludwich, 2022-12-22:
@reyntjensw do you have a power sensor?
Don´t want to push something i can´t verify.

andiwirz, 2022-12-22:
Yes - I have a Power Sensor

reyntjensw, 2022-12-22:
> @reyntjensw do you have a power sensor? Don´t want to push something i can´t verify.

I have one as well.

andiwirz, 2022-12-22:
I'm not a pro - but would this add the two values?

https://github.com/reyntjensw/Homey-FusionSolar/blob/d51b7cb0690cb9b2dfe72b626ab70c3bbc1e88ed/drivers/luna/device.js

reyntjensw, 2022-12-22:
@andiwirz can you test the change ?
In my case the values are not there, this is my output 
`{
  pv2_u: 0,
  pv4_u: 0,
  pv22_i: 0,
  power_factor: 0,
  pv6_u: 0,
  pv24_i: 0,
  mppt_total_cap: 11847.66,
  pv8_u: 0,
  open_time: 1671698966000,
  pv22_u: 0,
  a_i: 0.197,
  pv24_u: 0,
  mppt_9_cap: 0,
  c_i: 0.239,
  pv20_u: 0,
  pv19_u: 0,
  pv15_u: 0,
  a_u: 228.9,
  reactive_power: 0.001,
  pv17_u: 0,
  c_u: 233.1,
  mppt_8_cap: 0,
  pv20_i: 0,
  pv15_i: 0,
  pv17_i: 0,
  efficiency: 100,
  pv11_i: 0,
  pv13_i: 0,
  pv11_u: 0,
  mppt_power: 0,
  pv13_u: 0,
  run_state: 1,
  close_time: 'N/A',
  pv19_i: 0,
  mppt_7_cap: 0,
  mppt_5_cap: 0,
  pv2_i: 0,
  pv4_i: 0,
  active_power: 0,
  pv6_i: 0,
  pv8_i: 0,
  mppt_6_cap: 0,
  pv1_u: 0,
  pv3_u: 0,
  pv23_i: 0,
  pv5_u: 0,
  pv7_u: 0,
  pv23_u: 0,
  inverter_state: 512,
  pv9_u: 0,
  total_cap: 10103.86,
  mppt_3_cap: 0,
  b_i: 0.221,
  pv21_u: 0,
  mppt_10_cap: 0,
  pv16_u: 0,
  pv18_u: 0,
  temperature: 34.5,
  bc_u: 397.5,
  b_u: 224.3,
  pv21_i: 0,
  elec_freq: 49.96,
  mppt_4_cap: 0,
  pv16_i: 0,
  pv18_i: 0,
  day_cap: 2.31,
  pv12_i: 0,
  pv14_i: 0,
  pv12_u: 0,
  mppt_1_cap: 5569.98,
  pv14_u: 0,
  pv10_u: 0,
  pv1_i: 0,
  pv3_i: 0,
  mppt_2_cap: 6277.68,
  pv5_i: 0,
  ab_u: 393.2,
  ca_u: 397.9,
  pv7_i: 0,
  pv10_i: 0,
  pv9_i: 0
}`

andiwirz, 2022-12-22:
How did you get the data in this format?
Updated driver.compose.json as well

ludwich, 2022-12-22:
> @andiwirz can you test the change ? In my case the values are not there, this is my output `{ pv2_u: 0, pv4_u: 0, pv22_i: 0, power_factor: 0, pv6_u: 0, pv24_i: 0, mppt_total_cap: 11847.66, pv8_u: 0, [...]

This is not from the power sensor right?,

ludwich, 2023-01-02:
@andiwirz close this if you are happy in test.

closed 2023-01-02 by andiwirz