reyntjensw/Homey-FusionSolar #29: Battery data not working

Vaggan71, 2022-12-21:
Hello,
I have battery for my system that stopped giving info since version 1.0.34.
Even version 1.0.35 does not work; in the code it looks like you just changed that the battery id must not be null or empty.

It is stuck at 65% and Charge battery -0.38.

Is there any way I can install version 1.0.33 again; it worked perfectly for me.

reyntjensw, 2022-12-21:
@ludwich can you take a look at this please?

ludwich, 2022-12-21:
@Vaggan71 I don´t have a battery myself and the logic for setting these values are the same as 1.0.33 but if you send my your login credentials i could look at your setup. ludwig.hallgren@gmail.com

reyntjensw, 2022-12-21:
@Vaggan71 the new build is live, can you please check version 1.0.36?

closed 2022-12-21 by reyntjensw

Vaggan71, 2022-12-21:
Thx, its working.

PR #30 (2022-12-21): removed typo on battery
Should resolve #29