reyntjensw/Homey-FusionSolar #39: Cannot read property 'name' of undefined

HerrLindberg, 2023-01-18:
Hi!

Same problem as the others.

Here is my diagnostic...

cd23ca43-f3bf-4e71-b4a8-c99c753d49f2

Regards Tobias

reyntjensw, 2023-01-18:
good afternoon @HerrLindberg , you cannot use your email address to login into this system.
You must request an API user at your installer.

HerrLindberg, 2023-01-25:
Still same problems.
Now with new user name...
Here is my diagnostic...

f0dbb750-e280-4958-af80-24b8a3-978efe

/L

reyntjensw, 2023-01-25:
@HerrLindberg did you receive the username from your installer?
I do see that it is not returning systems.

In some cases the api access is not working because of a lack of permissions.

HerrLindberg, 2023-01-25:
Hi!

Yes!

I have new username without any strange symbols etc.

What kind of permissions?

I´m the owner and do n ot have the installer rights...

/L

HerrLindberg, 2023-01-25:
ps. I´m on this server...

https://region02eu5.fusionsolar.huawei.com/

HerrLindberg, 2023-01-25:
https://region02eu5.fusionsolar.huawei.com/pvmswebsite/nologin/assets/build/index.html#/kiosk?kk=n7aKmJoCjbdM88uMCyoAE7A3DxyCq1no

reyntjensw, 2023-01-27:
@HerrLindberg I do see that there are no systems connected to the api account.
Please ask your installer to double check.

reyntjensw, 2023-04-12:
I have released a new version, can you please try again?

HerrLindberg, 2023-04-12:
Still the same!

98fa632a-6247-4c4a-8d2d-838d7e524d77

reyntjensw, 2023-04-12:
Good evening
 
I still see the error  { failCode: 305, immediately: true, message: 'USER_MUST_RELOGIN'
 
Have you contacted Huawei about this?
 
Wouter

reyntjensw, 2023-05-02:
In my case my account expired, they had to create a new account for me

closed 2023-05-09 by reyntjensw

HerrLindberg, 2023-06-18:
Problem solved with new version!

Thanks for everything!

/T