reyntjensw/Homey-FusionSolar #40: Missing to recieve all data.

Solsind, 2023-01-21:
Hey.
I really love the idea to get my solar in homey. It is done and the api is created and works fine. My problem is that the data only shows in yield today and total, and maybe one more. Rest of the fields are empty and show no result.? What do I do wrong.? Do you have plans for making it possible to make flows where I can start charging from grid when the price are low, and maybe sell from battery when the price is high.?

joecool69, 2023-01-22:
I have the same issue..  would like to read the power

reyntjensw, 2023-01-22:
> Hey. I really love the idea to get my solar in homey. It is done and the api is created and works fine. My problem is that the data only shows in yield today and total, and maybe one more. Rest of [...]

Can you provide me with a screenshot please?
I'd like to know how many people are interested in buying or selling power to the grid, this is a rather big change

Solsind, 2023-01-23:
[image]
[image]

Solsind, 2023-01-23:
Here in the north off Europe, the sun is gone 4 pm in the winter so we buy from grid to store on batteries when the price is lowest. It is not always the same time of day, so it is manual every day, where I look op in spot price to se where the best time to buy is. Could be cool if I could use flows to do that job, maybe with power by the hour.

ludwich, 2023-01-23:
@Solsind do you know how to use postman? Would be good to see your returned values from devlist.

Solsind, 2023-01-23:
No im sorry i dont know how to do that

trapeze78, 2023-01-28:
Hi! I would love to have this feature as well and I am aware of at least one more user outside of this forum with the same wish.

Keep up the good work!

Marten

reyntjensw, 2023-01-28:
@trapeze78 if you would have the postman calls, that would already help a lot!
Looking forward to it.

trapeze78, 2023-01-28:
Thanks! I am just awaiting my API user from the Huawei team, following your instructions but they have been a little reluctant so far even though I have an installer account in place.

I will keep you posted!

joecool69, 2023-01-28:
Hi, I can see my Yield values in Homey, but not the rest like Power [W].  I can do some tests with Postman if you send me the GET

ludwich, 2023-01-29:
https://forum.huawei.com/enterprise/en/communicate-with-fusionsolar-through-an-openapi-account/thread/591478-100027

poblom, 2023-03-04:
I made some tests with Postman and am able to login, "getStationList", "getStationRealKpi" and "getKpiStationHour". But trying "getKpiStationHour" (where it starts to be interesting) is giving failCode 20012 (whatever that means). Could this be a permission error? Could it be that my login credentials are some sort of guest account with limited access to data?
{
    "data": null,
    "failCode": 20012,
    "message": null,
    "params": {
        "currentTime": 1677959095539,
        "stationCodes": "NE=XXXXXXXX"
    },
    "success": false
}

poblom, 2023-03-04:
I made some tests with Postman and am able to login, "getStationList" and "getStationRealKpi". But trying "getKpiStationHour" (where it starts to be interesting) is giving failCode 20012 (whatever that means). Could this be a permission error? Could it be that my login credentials are some sort of guest account with limited access to data?
{
"data": null,
"failCode": 20012,
"message": null,
"params": {
"currentTime": 1677959095539,
"stationCodes": "NE=XXXXXXXX"
},
"success": false
}

reyntjensw, 2023-03-04:
@poblom I think there is something wrong with your account.
[This link](https://developer.huawei.com/consumer/en/doc/development/HMSCore-Guides/open-platform-error-0000001053869182) shows you what the error code is about.

poblom, 2023-03-04:
Not much wiser from that list I'm afraid - "The format of response_type is incorrect. It has a fixed value of code". Perhaps it's "stationCodes" value that is the culprit with a strange format. But ok, I will have another word with my installer to see if they can help. Do I need an "installer account"?

reyntjensw, 2023-03-04:
You don't need an installer account, I would try to contact Huawei support and ask them what they think.

reyntjensw, 2023-04-13:
@poblom can you please try again, there is a new version available.

poblom, 2023-04-13:
@reyntjensw  Installed 1.0.48, seems to work great! Thank You!

P.S Only one little annoyance - wrong unit for Installed capacity

reyntjensw, 2023-04-13:
@poblom Good to hear.

What is the issue with the installed capacity? It should be kWp https://github.com/reyntjensw/Homey-FusionSolar/blob/main/drivers/luna/driver.compose.json#L51

poblom, 2023-04-13:
@reyntjensw  Yes, but in the app it's kWh!

reyntjensw, 2023-04-13:
@poblom can you try to remove the app and add your installation again?

poblom, 2023-04-13:
@reyntjensw  Sorry, but that was not a very good idea. Trying to get a new device only give "null is not an object (evaluating 'device.length')"

reyntjensw, 2023-04-13:
> @reyntjensw Sorry, but that was not a very good idea. Trying to get a new device only give "null is not an object (evaluating 'device.length')"

Can you send me a diagnostics report?

poblom, 2023-04-13:
Tried byt got a red label saying "Rate limited"???  Or did you get a report?? Will give it another try

reyntjensw, 2023-04-13:
Did not get anything

poblom, 2023-04-13:
This is strange, I am unable to even send a report. I guess I will have to try again tomorrow.

poblom, 2023-04-14:
No, I am still not able to add a device, same message as before,  "null is not an object (evaluating 'device.length’)”.

Sending a diagnostic report is not possible. Get a message ”Rate Limited” and no report is created! (is that a Homey message?)

Do You have an earlier version that i can download and try? I have had a working device for some time with all the values that is possible without the smart meter. I am no consol master but am able to do a CLI install.

PO

> 
> 
> Did not get anything

P-O Blomstrand
***@***.***

Edared Lövås
51996 Fotskäl

0705-724474

Treat your password like your toothbrush.  Don't let anybody else use it, and get a new one every six months.

reyntjensw, 2023-05-02:
@poblom I need the diagnostics report in order to troubleshoot, can you make a Homey support case for this?

poblom, 2023-05-02:
@reyntjensw  I fully understand that you need a diagnostics report. The problem persists thou, that is I am somehow "Rate Limited"

Yes, I already made a support case at Athom on this issue about a week ago but no reply! What to do more??

I tried with Postman and saw my plant properly listed. So login and the next step worked fine.

reyntjensw, 2023-05-02:
@poblom  does your installation have a power meter?
What do you get when you do a getDevList call : https://forum.huawei.com/enterprise/en/communicate-with-fusionsolar-through-an-openapi-account/thread/591478-100027

If that is not showing any powermeter then that is causing it.

Please post the outcome so I can have a look

poblom, 2023-05-02:
@reyntjensw No, I don't have a power meter, never had. Still want me to try out getDevList?

reyntjensw, 2023-05-03:
From what I know, if there is no power meter, I don't think we can read other values.
If people have the homeassistant app where you see more values without a powermeter, let me know and please send me a screenshot

poblom, 2023-05-04:
@reyntjensw 
Finally managed to create a diagnostics report: b10ecc32-c80b-4ee6-8495-70d3a1790765

reyntjensw, 2023-05-04:
@poblom this is the error I'm getting, I think your API access does not grant you access to list the devices
The below error is a Huawei error, so I would suggest to contact the supplier

`{ data: null, failCode: 401, message: 'Invalid access to current interface!', params: {}, success: false }`

poblom, 2023-05-04:
@reyntjensw 
I find this utterly strange. How could my access rights change. All I have done on my part is remove a nicely working device from Homey (a big mistake by the way). 
But ok, I will do some tests with Postman and see what that bings. And eventually contact my installer and ask if they changed anything. But why would they....
/PO

reyntjensw, 2023-05-04:
@poblom can you please test the calls in postman, if that gives another result I'll investigate.

poblom, 2023-05-07:
@reyntjensw 
For the time being, at least, I won't bother trying to get it working on my Homey23. I started my old Homey16 the other day (It's been off for a month since I got my new Homey) and found that I had a working device there. So, I will let it running for now. Just a few clicks away. Obviously my credentials are ok!
Below see the readings I get.
[image]
[image]

Tschurl, 2023-05-09:
This year (May 2023) and I have a DTSU-666H power meter for Huawei

reyntjensw, 2023-05-10:
@Tschurl please install the new test version

reyntjensw, 2023-05-10:
@poblom please install the new test version

Tschurl, 2023-05-11:
> @Tschurl please install the new test version

Hello, thanks! I installed the new version and it worked. But after a couple of minutes the message _"Error retrieving data (TypeError: Cannot convert undefined or null to object)"_ appears. And after some more minutes the message disappears and the values are showed again.

reyntjensw, 2023-05-20:
@Tschurl and @poblom can you please read this article and try the web method please? https://community.homey.app/t/app-pro-huawei-solar-smart-energy-for-a-better-life/61745/152?u=reyntjensw

Tschurl, 2023-05-21:
@reyntjensw I tried but I cannot connect like described here https://community.homey.app/t/app-pro-huawei-solar-smart-energy-for-a-better-life/61745/154?u=reyntjensw

reyntjensw, 2023-05-21:
@Tschurl I assume you are not in the region01eu5 region, can you please verify?
I'm working on a fix for this.

Tschurl, 2023-05-21:
@reyntjensw My webaccount - Login is here eu5.fusionsolar.huawei.com/unisso/login.action?service=https%3A%2F%2Fregion01eu5.fusionsolar.huawei.com%2Funisess%2Fv1%2Fauth%3Fservice%3D%252Fpvmswebsite%252Fassets%252Fbuild%252Findex.html%2523%252Fview%252Fstation%252FNE%253D35905345%252Foverview&decision=1
so I think I am in region01eu5

reyntjensw, 2023-05-21:
@Tschurl can you also give me the url when you are logged in? This is only your login url.
Can you also try to remove the app and reinstall the test version please?

Tschurl, 2023-05-21:
@reyntjensw https://region01eu5.fusionsolar.huawei.com/pvmswebsite/assets/build/index.html#/view/station/NE=35905345/overview

I will reinstall and give you feedback in about 3 hours due to I am not at home :)

Tschurl, 2023-05-21:
[image]
If I login with the new web-feature I get this and no device is installed

Tschurl, 2023-05-21:
@reyntjensw adding a device after2023 nearly working 😅
[image]

reyntjensw, 2023-05-28:
@Tschurl please use the web version, this should work

closed 2023-05-28 by reyntjensw

Tschurl, 2023-05-29:
Hi @reyntjensw : the web-version is working, but I am missing a real important parameter for me: **import/export to grid**
[image] like on the other version. This parameter is important for me due to I heat water if the PV produces more energy than I consume.