reyntjensw/Homey-FusionSolar #45: Cannot read property 'name' of undefined

Churchwood, 2023-03-12:
Hey,
Sorry to ask same question as many others. I am installer and created accounts with all privileges checked off. Do you have an idea why it still does not work?

reyntjensw, 2023-03-12:
No problem, so the issue is in most cases that the api is not returning any systems.
Follow the login guide on the site of Huawei https://forum.huawei.com/enterprise/en/communicate-with-fusionsolar-through-an-openapi-account/thread/591478-100027

Step 1 login and step 5 are important.
If this is working, you are fine, if step 5 is not returning anything, then there is a issue.

But I cannot tell you how to solve this.

reyntjensw, 2023-03-12:
> Hey, Sorry to ask same question as many others. I am installer and created accounts with all privileges checked off. Do you have an idea why it still does not work?

No problem, so the issue is in most cases that the api is not returning any systems.
Follow the login guide on the site of Huawei https://forum.huawei.com/enterprise/en/communicate-with-fusionsolar-through-an-openapi-account/thread/591478-100027

Step 1 login and step 5 are important.
If this is working, you are fine, if step 5 is not returning anything, then there is a issue.

But I cannot tell you how to solve this.

Churchwood, 2023-03-12:
It stops when it comes to getStationList via Postman. Login works fine. Just wrote Huawei to see if they can help. Will keep you posted

reyntjensw, 2023-03-12:
> It stops when it comes to getStationList via Postman. Login works fine. Just wrote Huawei to see if they can help. Will keep you posted

Correct, that is what I see most of the time.

Churchwood, 2023-03-12:
I see but frustrating to be admin and not able to find out what is wrong. The users have all access available

reyntjensw, 2023-03-12:
> I see but frustrating to be admin and not able to find out what is wrong. The users have all access available

I understand completely, the issue I have is that I don't know what Huawei does to fix it.
Otherwise I could point you in a direction.

Churchwood, 2023-03-12:
I will keep you posted when I receive feedback from them.

Danne1975, 2023-03-13:
I have the same issue with me login, have been in contact with my installer and according to them all is set up correct.

reyntjensw, 2023-03-13:
@Danne1975 Please check this comment https://community.homey.app/t/huawei-solar/61745/96?u=reyntjensw

Danne1975, 2023-03-15:
Morning,

Yesterday evening I tried my API log in on my newly installed home
assistent and it worked 👍. So the log in works fine so there is something
connected with Homey that is not correct.

Kind Regards
Johan

reyntjensw, 2023-03-15:
> Morning, Yesterday evening I tried my API log in on my newly installed home assistent and it worked 👍. So the log in works fine so there is something connected with Homey that is not correct. Kind Regards Johan Den mån 13 mars 2023 20:34Wouter Reyntjens ***@***.***> skrev:
> […](#)

Hi Johan

Can you send me a diagnostics report from Homey please?

Danne1975, 2023-03-15:
Hi, yes no problem.

Please guide me so I get it right.

1. Try to ""log in" witch will fail and I get attached screenshot.

2. Should then create diagnostics report and send in Homey?

Best Regards
Johan

reyntjensw, 2023-03-15:
Your steps are correct.
Please follow these steps https://community.homey.app/t/huawei-solar/61745/70?u=reyntjensw

Danne1975, 2023-03-15:
Will do now. Apperanty Homey can't read the devices. In HA it was 6 devices
and around 200 different values from my API.

Danne1975, 2023-03-15:
Another observation, sometimes I do not get the message can't read devices.
I'm asked to choose witch units to add but there are no units to choose
from (complete blank) then I push continue. Since I have chosen any devices
nothing is visible under units/devices in the Homey app

Danne1975, 2023-03-15:
Here is the Report

e0e6ebef-ee77-4989-b45b-4f3aac09da3d

Hope you can detect the error

Danne1975, 2023-03-15:
Did you ask for my API credentials?

reyntjensw, 2023-03-15:
> […](#)

yes I did

Danne1975, 2023-03-15:
Ok, must admit always a little bit careful of given this out...but do you
think it will solve my issue?

Churchwood, 2023-03-15:
Hello Gents,
Happy to see that this issue gets attention. Super great. Thx

Received these mails from Huawei yesterday. I am busy at work today but it might be useful to you.

All the best
René

Best regards

René Kirkeskov Olesen
Director

Cell. +45 53 6000 43

ACAX
Frydensbergvej 20
3660 Stenløse
Denmark

reyntjensw, 2023-03-15:
> Hello Gents, Happy to see that this issue gets attention. Super great. Thx Received these mails from Huawei yesterday. I am busy at work today but it might be useful to you. All the best René Best [...]

Good evening, We cannot see the mails attached.
Could you paste the text please?

Churchwood, 2023-03-15:
> Hello,
>
> Please be advised that we created related SR NO. 8710505
>
> We will come back with a resolution as soon as possible.
>
> Best regards,
>
> Alex Strejaru
>
> Customer Care Representative
> CCR Team Leader – Madalina Condel   Email : ***@***.******@***.***>
> Huawei Enterprise Solar Inverter Global Service Center will assist you 24x7x365
> Home : https://solar.huawei.com/eu
>
> This e-mail and its attachments contain confidential information from HUAWEI, which is intended only for the person or entity whose address is listed above. Any use of the information contained here in any way (including, but not limited to, total or partial disclosure, reproduction, or dissemination) by persons other than the intended recipient(s) is prohibited. If you receive this e-mail in error, please notify the sender by phone or email immediately and delete it!
> Note: In compliance with the provisions of the Federal Law of Protection of Personal Data in Possession of Individuals, HUAWEI makes available its Privacy Notice, which may be consulted at https://www.huawei.com/en/privacy-policy
>
>
> Hello again,
> I am administrator. I have tried several times to follow the guide you provided. Anyways I am not able to login via Homey APP nor Postman following this guide
> https://forum.huawei.com/enterprise/en/communicate-with-fusionsolar-through-an-openapi-account/thread/591478-100027
>
> I created 3 users to test if it works with different settings. They are able to login and generate a token but then it stops. When trying next call I receive USER_MUST_RELOGIN.
>
> Can you please provide some advice?

Sure. Here go the text:

Mail:
Hello mister Rene,

Sorry for the discomfort! There is a small problem that infiltrated that procedure.
Still, the correct syntax is provided by the procedure that is attached below.
Please let us know if we can assist further regarding the subject.

Regards,
Nicolae Stan
Service Helpdesk Engineer

Team Leader Helpdesk – CORINA PIRVU  Email : ***@***.******@***.***>
Huawei Enterprise Solar Inverter Global Service Center will assist you 24x7x365
Home : https://solar.huawei.com/eu
***@***.***<https://support.huawei.com/enterprise/en/index.html>
***@***.***<https://forum.huawei.com/enterprise/en/Network-Energy/forum/100027?typeid=2313>
***@***.***<https://e.huawei.com/en/service-hotline-query>
***@***.***<http://community.solar.huawei.com/en/home>
***@***.***<https://support.huawei.com/enrobot/es/index.html>
***@***.***<https://support.huawei.com/enterprise/en/category/hybrid-power-pid-22318399>
***@***.******@***.***>

This e-mail and its attachments contain confidential information from HUAWEI, which is intended only for the person or entity whose address is listed above. Any use of the information contained here in any way (including, but not limited to, total or partial disclosure, reproduction, or dissemination) by persons other than the intended recipient(s) is prohibited. If you receive this e-mail in error, please notify the sender by phone or email immediately and delete it!
Note: In compliance with the provisions of the Federal Law of Protection of Personal Data in Possession of Individuals, HUAWEI makes available its Privacy Notice, which may be consulted at https://www.huawei.com/en/privacy-policy

Hello,

In order for the API account to be created you need to contact the company administrator and ask him to do the API account like in the procedure attached.
Please let us know if we can assist further regarding the subject.

Regards,
Nicolae Stan
Service Helpdesk Engineer

Team Leader Helpdesk – CORINA PIRVU  Email : ***@***.******@***.***>
Huawei Enterprise Solar Inverter Global Service Center will assist you 24x7x365
Home : https://solar.huawei.com/eu

This e-mail and its attachments contain confidential information from HUAWEI, which is intended only for the person or entity whose address is listed above. Any use of the information contained here in any way (including, but not limited to, total or partial disclosure, reproduction, or dissemination) by persons other than the intended recipient(s) is prohibited. If you receive this e-mail in error, please notify the sender by phone or email immediately and delete it!
Note: In compliance with the provisions of the Federal Law of Protection of Personal Data in Possession of Individuals, HUAWEI makes available its Privacy Notice, which may be consulted at https://www.huawei.com/en/privacy-policy

【Notice】 Please keep the SR No. 8624351 in the subject, otherwise our engineer can’t receive your email.If the engineer doesn't receive a reply or feedback in maximum 48 hours the case will be closed automatically.
Hello,
       We have received your request and we have created the new service request 8624351.
       We will come back to you ASAP with a solution.
       Thank you in advance!
Andreea Tache
Customer Care Representative
CCR Team Leader – Madalina Condel   Email : ***@***.******@***.***>
Thank You for Contacting Huawei Solar Inverter Global Service Center Romania
Huawei Enterprise Technical Assistant Center will assist you 24x7
E-mail: ***@***.******@***.***>
24*7 Hotline:
Europe +80 03 38 88 888  (except below countries)
Croatia/Latvia/Bulgaria +80 07 77 78 899
Iceland 800201866 Slovakia +40 31 22 61 915 Moldova 080061466
Serbia 800190899 Lithuania 880031499 Romania 0800400897
Australia 1800046639 Canada +1 855 482 9343 New Zealand 080061466
Phone Support for more regions: http://e.huawei.com/en/service-hotline
Address: Barbu Vacarescu Blvd, No. 201, GLOBAL WORTH Building ,14th floor, 2nd district, 020276, Bucharest Romania
http://solar.huawei.com/

Hi
I would like to imtegrate my Huawei inverter to my Homey smarthome. For that I understand that you can help me with the API integration
Please find my serial number below
Inhope for your support

Sn TA2270326361

------------------------------------------------------
René Kirkeskov Olesen
------------------------------------------------------

Mob: +45 53 60 00 43

Churchwood, 2023-03-15:
[北向指南SmartPVMS V500R007C00 NBI Reference.docx](https://github.com/reyntjensw/Homey-FusionSolar/files/10984190/SmartPVMS.V500R007C00.NBI.Reference.docx)

reyntjensw, 2023-03-15:
> Sure. Here go the text: Mail: Hello mister Rene, Sorry for the discomfort! There is a small problem that infiltrated that procedure. Still, the correct syntax is provided by the procedure that is [...]

That is not the answer I was hoping from. I would reply to them again with a screenshot of the error you are facing.

Churchwood, 2023-03-15:
I fully agree with you. Already asked them ;-). Answer is useless. However I curious about the V5 & V7 settings. In my admin portal I am not able to pick V7 settings.

Churchwood, 2023-03-18:
got feedback once again. No news except that I was to use Postman. I tried once again. Exactly same result. @reyntjensw  can you test with my credentials?

reyntjensw, 2023-03-18:
> got feedback once again. No news except that I was to use Postman. I tried once again. Exactly same result. @reyntjensw can you test with my credentials?

The steps I do are the same as described in this document https://forum.huawei.com/enterprise/en/communicate-with-fusionsolar-through-an-openapi-account/thread/591478-100027

Step 1 till 7 is what I do

reyntjensw, 2023-04-12:
I have released a new version, can you please try again?

Churchwood, 2023-04-12:
Hey
I am home tomorrow and will check. Home assistant works ok
Rgds

Best regards

René Kirkeskov Olesen

Director

Cell. +45 53 6000 43

ACAX Aps

Blodhøjvej 16

4040 Jyllinge

Denmark

Churchwood, 2023-04-19:
Hello again,
Now I have access and are able to create a new device. Unfortunately it only works for a short while. Then data is not shown anymore. FYI I have no issues on HomeAssistant. And of course I would love to get Homey behave so I can keep using one platform

Best regards

René Kirkeskov Olesen
Partner
***@***.***
***@***.*** +4553600043 <tel:+4553600043>   ***@***.***  www.viscongroup.dk <https://www.viscongroup.dk/>   ***@***.***  ***@***.******@***.***>
***@***.*** Flaesketorvet 68, 1711 Copenhagen V, Denmark<https://maps.google.com/?q=Flaesketorvet%2068,%201711%20Copenhagen%20V,%20Denmark>

reyntjensw, 2023-04-20:
@Churchwood please hang on, I have issues with my Huawei api account and I'm not able to do any testing. When this is resolved, I'll fix the issue

reyntjensw, 2023-05-02:
@Churchwood can you please give the test version a try?

Churchwood, 2023-05-03:
Hey,
Tried once again. Following error: Cannot read property ‘name’ of undefined

Best regards

René Kirkeskov Olesen
Partner
***@***.***
***@***.*** +4553600043 <tel:+4553600043>   ***@***.***  www.viscongroup.dk <https://www.viscongroup.dk/>   ***@***.***  ***@***.******@***.***>
***@***.*** Flaesketorvet 68, 1711 Copenhagen V, Denmark<https://maps.google.com/?q=Flaesketorvet%2068,%201711%20Copenhagen%20V,%20Denmark>

reyntjensw, 2023-05-20:
@Churchwood can you please read this article and try the web method please? https://community.homey.app/t/app-pro-huawei-solar-smart-energy-for-a-better-life/61745/152?u=reyntjensw

Churchwood, 2023-05-21:
Hello again. Tried this evening. Same reply as with the old way. Still it does work with Home Assistant
All the best

R

René Kirkeskov Olesen
Partner
***@***.***

***@***.*** +45 53 60 00 43  ***@***.***  www.viscongroup.dk <https://www.viscongroup.dk/>   ***@***.***  ***@***.******@***.***>
***@***.*** Flaesketorvet 68, 1711 Copenhagen V, Denmark<https://goo.gl/maps/UGwPrG8FTmchXkHt9> ***@***.***  +45 70 770 771

reyntjensw, 2023-05-21:
@Churchwood the web method is using the same way that is being used in Home Assistant.
Could you share with me your credentials so I can test please? It can be using a DM on the Homey community

Churchwood, 2023-05-21:
Hey
Please try this
Churchwood_homey
Pw
homLove2017!

Best regards

René Kirkeskov Olesen

Director

Cell. +45 53 6000 43

ACAX Aps

Blodhøjvej 16

4040 Jyllinge

Denmark

reyntjensw, 2023-05-21:
@Churchwood I have removed your credentials from Github.
I found the issue, can you please install 2.0.5 and try again?

reyntjensw, 2023-05-21:
@Churchwood the provided credentials did not work on the Huawei website.

closed 2023-05-28 by reyntjensw