reyntjensw/Homey-FusionSolar #51: TypeError: Cannot convert undefined or null to object

turbolover, 2023-04-15:
My homey icon shows the red triangle and when I click the icon I get the 
"TypeError: Cannot convert undefined or null to object"

I removed my plant and readded my plant with not effect. I restarted the application with no effect. Any ideas?

reyntjensw, 2023-04-15:
Can you send me a diag report please?

turbolover, 2023-04-16:
c0ad1984-1749-4ca9-89bc-bdd1b562f6c9

regards! matthias

reyntjensw, 2023-04-16:
Hi @turbolover  I see this error

`{
    "data": "ACCESS_FREQUENCY_IS_TOO_HIGH",
    "failCode": 407,
    "params": null,
    "success": false
}`

I would suggest waiting a bit and trying again later on

turbolover, 2023-04-16:
rebooted homey and app, no effect, haven't been able to get it working
again, same error for days . regards! matthias

reyntjensw, 2023-05-02:
@turbolover any updates?

turbolover, 2023-05-05:
Hi Wouter, no, no changes. My electrician had some issues sind 26.4 where
Fusion Solar can't see my plant anymore, so he has to fix this first.
Regards, Matthias

reyntjensw, 2023-05-20:
@turbolover can you please read this article and try the web method please? https://community.homey.app/t/app-pro-huawei-solar-smart-energy-for-a-better-life/61745/152?u=reyntjensw

turbolover, 2023-05-22:
Hi Wouter, I don't see the third version yet. My user is
https://region01eu5.fusionsolar.huawei.com/  Regards, Matthias

reyntjensw, 2023-05-22:
Can you please us the test version as described in the homey article? https://homey.app/nl-be/app/com.huawei.solar/Huawei-Solar/test/

turbolover, 2023-05-22:
Hi Wouter
Installed the Test Version. Connects fine, no time out, no error message.

a) icons switched
[image: image.png][image: image.png]

To me, it looks like the EXPORT and IMPORT are switched. My Load is under
1kW and I export up to 13kW. Or maybe it was just a temporary spike. But
maybe check.

b)
If you disable the battery in the settings, it would be perfect to NOT
display the icons in the overview above. Minor feedback.

Regards, Matthias

reyntjensw, 2023-05-28:
@turbolover I do not see the images you posted. Could you share them please?

turbolover, 2023-05-29:
Hi Wouter

I am running "Experimental 2.0.8"
https://homey.app/nl-be/app/com.huawei.solar/Huawei-Solar/test/

a)
I think, some of  the icons don't match the values in my Fusion Solar.
Your icon TOTAL POWER is showing my Fusion Solar TOTAL YIELD (over time)
Your icon ACTIVE USAGE and POWER and PV yield have the same value: my
current usageshowing my Fusion Solar
Your icon EXPORT TO GRID has also an unknown value, it's in kW and shows
52kW where my plant is 14kW so it cannot be.

There are some mismatches I think. The numbers don't add up
LOAD+EXPORT TO GRID = TOTAL POWER.  YIELD is mostly used as an over time
expression

b)
If you disable the battery in the settings, it would be perfect to NOT
display the battery icons in the overview above. Minor feedback.

c)
Today I got the error message "Cannot read properties of undefined (reading
'SOC'), I restarted the app.
Regards, Matthias

Regard, Matthias