reyntjensw/Homey-FusionSolar #52: Cannot read property 'name' of undefined, and tried all guide and Huawei support

GenAladeen, 2023-04-19:
Hi, experiance the same issue like many other here before me.
I have tried Postman and I can use my API login but fails when I ask for StationList.
Been in contact with Huawei support four times and even got then to confirm that my plant is connected to my account. (Which contradicts the previous sentence)
Yet it says ”cannot read proporty name of undefined”
Starting to run out of options except to get the insights from the developer himself.
Can you send me your wisdom of what I can do next?
Have all my automation running on Homey, all but one, my solar power.

reyntjensw, 2023-04-25:
Good afternoon, do you have an installer who you can contact?
They can create the api accounts as well.

I'll suggest giving that a try

GenAladeen, 2023-04-26:
I got my account from an installer but they are just the middle man, they can’t create accounts themselves instead they get it from the retailer of Huawei and I can’t contact them directly. What I’m not sure of is if I’m doing something wrong in the step ask Postman for station list, that’s why I contacted Huawei support again to get them to confirm that my plant was linked to my API account, the answer was - yes. Still when I try to connect with Homey I get same problem like many others, same people who also claims Home Assistant can connect with the API login. Recently the app went from 1.0.47 to 1.0.50 so something must be creating problem. (Just saw a test for 1.0.52). I just want to work together with everyone else here looking for an answer to solve the problem of connecting to my rather expensive solar investment which I was hoping to have integrated with other functions in my house and current price from Tibber. Can I drop a crash-report for you to analyze and see where the coms fail. My plant is quite new and have the latest updates installed by the installation company. But like most companies they just install and update systems, have no clue of coms. They just sent it to their dealer who makes API account but might not either know what they do except trying to follow Huawei’s PDF guide on how to make accounts. As I see it the problem starts when Huawei stopped making API accounts and outsourced that to the installers. Now people like me are stuck in the middle, I get API account but don’t know if it’s correct setup, Huawei support gives short answers that it’s linked to my plant but does that mean everything else is flawless and correct?? And then when I try to connect with your app it doesn’t work. So you can understand it seem problematic since I don’t know in which direction to start, if I was the only one I might assume account problem but with Huawei saying it’s linked and so many others reporting the same problem it might be linked to what versions we use in our inverters. Shout out if I can send crash log.
Thanks for your work so far.

GenAladeen, 2023-04-28:
Update: After testing the 1.0.52 test version I can now add my inverter and get connection.
On the main dash board I can read some values, upper half seems to be receiving values, lower half doesn’t but that might just be since it’s a test version.

reyntjensw, 2023-05-02:
@GenAladeen does your installation have a power meter?
What do you get when you do a getDevList call : https://forum.huawei.com/enterprise/en/communicate-with-fusionsolar-through-an-openapi-account/thread/591478-100027

If that is not showing any powermeter then that is causing it.

Please post the outcome so I can have a look

reyntjensw, 2023-05-02:
I'm closing this one and moving on to #53

closed 2023-05-02 by reyntjensw