reyntjensw/Homey-FusionSolar #53: Cannot read property 'name' of undefined

sebastianyberg, 2023-04-26:
Hello, can u please check this out if this is problem with my account ? 
Diagnostics Report: d9182206-230b-46f3-ad70-e5846741c0e4

JorgenSeemann, 2023-04-28:
I have similar problem.
Removing and reinstalling didn't help :/

reyntjensw, 2023-04-28:
Can you try the test version please? https://homey.app/nl-be/app/com.huawei.solar/Huawei-Solar/test/

JorgenSeemann, 2023-04-28:
Hi Wouter.

Just tried, but same error when trying to add the inverter.
Tried both the prior 2023 and 2023, but same error (my installation is from
2021).

Wonder if Huawei has updated something...

PS. I do get a different error if im using incorrect user/pwd, so I think
this is ok 😉

Br
Jørgen

sebastianyberg, 2023-04-28:
Hello, now it's found it. Don't got any value jet but i will wait a while. Huawei are not superfast to update.. :)

GenAladeen, 2023-04-28:
Update: After testing the 1.0.52 test version I can now add my inverter and get connection.
On the main dash board I can read some values, upper half seems to be receiving values, lower half doesn’t but that might just be since it’s a test version.
(I have previously started the #52 thread and have a new inverter from 2023)

sebastianyberg, 2023-04-28:
I dont get any value.. :(

reyntjensw, 2023-05-02:
@GenAladeen does your installation have a power meter?
What do you get when you do a getDevList call : https://forum.huawei.com/enterprise/en/communicate-with-fusionsolar-through-an-openapi-account/thread/591478-100027

If that is not showing any powermeter then that is causing it.

Please post the outcome so I can have a look

JorgenSeemann, 2023-05-03:
I have created error report - 379a307f-5ed4-4567-9777-5280878f0c91

Doesnt seems to matter if i use the 1.0.50 or 1.9.52

Br

reyntjensw, 2023-05-20:
@JorgenSeemann can you please read this article and try the web method please? https://community.homey.app/t/app-pro-huawei-solar-smart-energy-for-a-better-life/61745/152?u=reyntjensw

JorgenSeemann, 2023-05-21:
Hello Wouter

I get a "Could not find that pair session" when I try to use web auth with
the 2.0.3 version, and so cannot add the unit :/

Using API (pre 2023) adds the unit, but instantly goes into error mode
"invalid capability: inverter_temperature

So sorry - no dice for me.

Br
Jørgen

JorgenSeemann, 2023-05-21:
Rebooted the app, now getting "cannot read property 'name' of undefined"
trying to use WEB auth.

reyntjensw, 2023-05-21:
Did you use the web credentials to access the website?
If so, can you give me the url you get when you login? Is it eu05region01?

JorgenSeemann, 2023-05-21:
I'm on region02eu5
Using HTTPS://eu5.fusionsolar.huawei.com
Web usr/pwd works fine on the website

JorgenSeemann, 2023-05-21:
b1a42512-cc42-4293-9317-4e36b6c8a4b2
Diag report

reyntjensw, 2023-05-21:
Can you please change the region in the app settings? Give that a moment and let me know if it works, it is possible that some urls are still hardcover on region 1 for testing

JorgenSeemann, 2023-05-21:
Can't :/.
When I try to save the change, it throws the error (inverter temperatur)
and doesn't save (thou it greys ot the tick)

reyntjensw, 2023-05-21:
Ok I need to make some changes to the app for this, hang on

JorgenSeemann, 2023-05-21:
Ok, let me know. 😀

reyntjensw, 2023-05-21:
@JorgenSeemann please give it a try https://homey.app/a/com.huawei.solar/test/

JorgenSeemann, 2023-05-21:
Sorry - same error with rhe API pre-2023 version. Cannot cannge server due
to the inverter_temperature error.

The WEB wont validate (even copying usr/pwd from my pwd manager to avoid
mistype)

So all in all no dice. Sorry :/

sebastianyberg, 2023-05-21:
I just got no data from api account and web i got cannot read property name of undefined.
I will like Modbus TCP id that an option, like one app in HA 😃

reyntjensw, 2023-05-21:
@sebastianyberg  could you please share with me your credentials via DM in the Homey community?
You are free to try and build the Modbus option 😉

JorgenSeemann, 2023-05-24:
Hi Wouter.

Ver 2.0.6:
API: Can now set server but looks like the server string isn't passed to
URL. Throws error about https://undefined.fusionsolar.huawei.com/...

Web auth: Seems like this is now working for me 😀

Cheers
Jørgen

closed 2023-05-28 by reyntjensw