reyntjensw/Homey-FusionSolar #54: Cannot read property ‘name’ of undefined

cobetec, 2023-05-01:
ID from homey report: 949f76-9c24-4c66-af1f-e06ef2f78b8f

reyntjensw, 2023-05-01:
Hi @cobetec 

I think your username or access to the system is incorrect

` [ManagerFlow] [FlowCardCondition][the-sun-is-still-shining] Warning: Run listener was already registered. Username : api.olloc.be { data: null, failCode: 401, message: 'Invalid access to current interface!', params: {}, success: false } systems : null Username : api.olloc.be { data: null, failCode: 401, message: 'Invalid access to current interface!', params: {}, success: false } systems : null Username : api.olloc.be { data: null, failCode: 401, message: 'Invalid access to current interface!', params: {}, success: false }`

reyntjensw, 2023-05-09:
Please take a look at this forum post https://forum.huawei.com/enterprise/en/v7-interface/thread/997229-100027

reyntjensw, 2023-05-20:
@cobetec can you please read this article and try the web method please? https://community.homey.app/t/app-pro-huawei-solar-smart-energy-for-a-better-life/61745/152?u=reyntjensw

closed 2023-05-28 by reyntjensw