reyntjensw/Homey-FusionSolar #57: Missing values and Installed Capacity

polisen79, 2023-05-09:
I'm using the Huawei Solar 1.0.52, test app.

With the test app I managed to add my Huawei solar plant, a newer 2023 installation.

However I'm missing some data, see picture. According to the installer my API account has all available rights.

If using Postman towards the API in the instructions on https://forum.huawei.com/enterprise/en/communicate-with-fusionsolar-through-an-openapi-account/thread/591478-100027

Step 3 is successfull but on step 4 (getStationList) I recieve following error...

{
    "data": null,
    "failCode": 401,
    "message": "Invalid access to current interface!",
    "params": {},
    "success": false
}

My installation doesn't have a battery but I do have a Huawei smartmeter. Every value is on the FusionSolar web app.

I alos noticed the value for Installed capacity is displayed wrong. The unit should be kWp, not kWh. Also the value should be devided by 1000. In my picture it's displayed as 19140kWh but should be instead 19.14kWp.

[image]

reyntjensw, 2023-05-09:
@polisen79 Please take a look at this forum post : https://forum.huawei.com/enterprise/en/v7-interface/thread/997229-100027

polisen79, 2023-05-11:
Whoohoo... With you latest version 1.0.56 I got the values I was missing, except battery which I don't have. Values are verified and are the same as displayed in FusionSolar webapp. Very good job.

The installed capacity is still displayed incorrect as above "I alos noticed the value for Installed capacity is displayed wrong. The unit should be kWp, not kWh. Also the value should be devided by 1000. In my picture it's displayed as 19140kWh but should be instead 19.14kWp."

[image]

polisen79, 2023-05-11:
Hmm, the app infopage is displayed correct for about 50sec, then displays error "Error retrieving data (TypeError: Cannot convert undefined or null to object)" for 50 sec then loops again.

polisen79, 2023-05-11:
I tried to verify the values displayed in the app. The first 6 seems correct.
The next 2 (active/negative energy) I can't find in the FusionSolar webapp. What are those called?
The next value Import/export grid I'm guessing if importing or exporting currently and how much? If that's so, then it's probably correct.
The next value, Load is incorrect. I think you have mistakenly linked it to "Active power" instead.
Inverter temp is correct.
Panel efficiency, no idea, are you calculating that or is it a actual value in FusionSolar app?

I can't get the data from PV Yield and Import/Export grid to match FusionSolar webapp, I'm guessing those are ok since the values are not taken from the same time. They differs some but not very much.

reyntjensw, 2023-05-20:
@polisen79  can you please read this article and try the web method please? https://community.homey.app/t/app-pro-huawei-solar-smart-energy-for-a-better-life/61745/152?u=reyntjensw

polisen79, 2023-05-21:
Yeah, I'm facing the "Cannot read properties of null (reading 'length')" when trying to add device with the web alternative but guessing it has to do with the region I have, region02eu5. I guees it's working fine with other region. Hopefully this region will work soon aswell. Thank you for all time you spend on this app. Your work is much appreciated,

reyntjensw, 2023-05-21:
@polisen79 a new version was pushed for testing https://homey.app/a/com.huawei.solar/test/
This should be fixed.

polisen79, 2023-05-21:
Nice, the latest version when I tried was 2.0.3. However 2.0.3 accepted my credentials and return error code as above. 2.0.4 doesn't like my credentials. I tried both region01eu5 since that was predefined in 2.0.3 and region02eu5 which is my region, same error message from Homey, my credentials not accepted. They work in FusionSolar webapp.

reyntjensw, 2023-05-21:
@polisen79  could you please share with me your credentials via DM in the Homey community?

closed 2023-05-28 by reyntjensw