reyntjensw/Homey-FusionSolar #58: Device Unavailable in version 1.0.56 (HP2023)

Vaggan71, 2023-05-11:
Stopped working after version 1.0.56.
Device Unavailable

I have HP2023.
/Magnus

Vaggan71, 2023-05-11:
Its working if I add a new "before 2023".
But all my flows is broken.

ludwich, 2023-05-11:
What do we have to re- add the system?

mackewigren, 2023-05-11:
Also device unavailable. Installed 2022.

reyntjensw, 2023-05-11:
> @reyntjensw same issue for me.
> 
> Can you push the code so i can see the diff.

The code is in this PR https://github.com/reyntjensw/Homey-FusionSolar/pull/55
The issue has to do with the backend server, I commented that one out and replaced it with the intl option.

I don't have my own account so I cannot test thoroughly.

Error 305 seems to be caused by the querying the wrong server

reyntjensw, 2023-05-11:
> Its working if I add a new "before 2023". But all my "solar" flows is broken.

That is normal when you remove a device, you can replace the broken device.

ludwich, 2023-05-11:
No we have to add a total new one so all flows and power by the hour data
is lost.

Vaggan71, 2023-05-12:
> > Its working if I add a new "before 2023". But all my "solar" flows is broken.
> 
> That is normal when you remove a device, you can replace the broken device.

I haven't removed any device, it was unavailable after the version 1.0.56 auto installed.

Vaggan71, 2023-05-15:
Now in version 1.0.57 its stoppad working agan.

Vaggan71, 2023-05-17:
@reyntjensw version 1.0.58 is working but my daily and total will not update.

reyntjensw, 2023-05-20:
@Vaggan71 can you please read this article and try the web method please? https://community.homey.app/t/app-pro-huawei-solar-smart-energy-for-a-better-life/61745/152?u=reyntjensw

mackewigren, 2023-05-21:
I tried the 2.0.3 experimental version. Had earlier experimental version before. 

Doesn’t work for me. Tried both “Before 2023” and “Web login”. 

With “Before 2023” the device is found but I get the red triangle. Then I try to change region/server and then get error about “Invalid capability inverter temperature”. Cannot get passed this so it doesn’t work. The device is added but with wrong settings. 

With “Web login” it states 1 device found but then get error about “Cannot read property ‘name’ of undefined” so the device is not added.

Vaggan71, 2023-05-21:
@reyntjensw 
I have tried version 2.0.3 but it doesn't work (before 2023). Web login does not work because I am in region02

reyntjensw, 2023-05-21:
@Vaggan71 can you please try the web login again? https://homey.app/a/com.huawei.solar/test/

mackewigren, 2023-05-21:
I also tried again. Could choose region first but then same message anyway.

Vaggan71, 2023-05-22:
@reyntjensw  In version 2.05 web login works, but battery does not work; see attached picture.

[image]

mackewigren, 2023-05-22:
Works fine (it seems) after restart also with battery

closed 2023-05-28 by reyntjensw