reyntjensw/Homey-FusionSolar #59: Server not defined in 1.0.57

ludwich, 2023-05-15:
Can please start to test your updates. Last update i hade to remove and readd so broke all data and flows and this Update it is not working only says server not defined.

reyntjensw, 2023-05-15:
@ludwich since a couple of weeks I have no api credentials so I am not able to test the releases.
I'm working on a way to use the website credentials instead of the openapi, because this is not maintainable for me

ludwich, 2023-05-15:
If you ping me and leave a test version for 1-2 weeks we can easily test it.

Tschurl, 2023-05-15:
If possible, I would like to help you?

8d7733e1-bdb4-4ca2-9bc0-89d13e0c3228

ludwich, 2023-05-16:
@reyntjensw Cant see the latest PR. If you push them I can have a look.

reyntjensw, 2023-05-16:
@ludwich the test version is available https://homey.app/a/com.huawei.solar/test/

Please let me know what de findings are.
This is the PR I'm working in https://github.com/reyntjensw/Homey-FusionSolar/pull/55

JorgenSeemann, 2023-05-16:
I confirm the .58 version has removed the "server is not defined " error.
On my system it seems to work as normal again.

Cheers
Jørgen

ludwich, 2023-05-16:
5 minut values are updated correctly. Waiting to see if daily gets updated

ludwich, 2023-05-16:
My daily and total will not update. Will check again tomorrow.

JorgenSeemann, 2023-05-16:
Looks like same for me.

JorgenSeemann, 2023-05-17:
Daily & monthly yield and total power
All showing last update is 2 days ago 😀

ludwich, 2023-05-17:
@reyntjensw Yes the daily & monthly and total power is not updating so we are not coming into that function for some reason.

ludwich, 2023-05-17:
@reyntjensw the bug is on line 97 in device.js. if (Object.entries(basicStatsObj).length !== 0)  this is going to be 0 in length and function wont be called. if you go basicStatsObj.length that is 1 and the function will be called.

Tschurl, 2023-05-17:
@reyntjensw same bug and another question: why are the values only every second minute available? 
[image]
[image]
[image]
[image]

BartDeWorm, 2023-05-20:
I have the same problem..
Can you help me to fix this? Or can I test something for you?
3e25f0ac-24e2-42a1-adb2-9325bd81d3f5

reyntjensw, 2023-05-20:
@BartDeWorm can you please read this article and try the web method please? https://community.homey.app/t/app-pro-huawei-solar-smart-energy-for-a-better-life/61745/152?u=reyntjensw

closed 2023-05-28 by reyntjensw

BartDeWorm, 2023-06-15:
Beste Wouter,

Heb het gevoel dat het probleem terug is..

Steeds een rood uitroepteken bij de omvormer en na herstart van de Homey
blijft die continu laden.

Kan je dit nogmaals nakijken aub?
Alvast bedankt!

Met vriendelijke groeten,
Bart De Worm