reyntjensw/Homey-FusionSolar #61: Device Unavailable

nissenhj, 2023-05-21:
Not getting it to work?  Neither with 1.0.57 nor 2.0.3

Getting Device Unavailable!
Invalid Capability: inverter_temperature, homey:app:com.huawei.solar

That's all I can come up with. 
API account works fine with BOVP.info on Windows computer
[image]

reyntjensw, 2023-05-21:
@nissenhj the api apps will go EOL soon, please try the web version using the test version https://homey.app/a/com.huawei.solar/test/

nissenhj, 2023-05-21:
> @nissenhj the api apps will go EOL soon, please try the web version using the test version https://homey.app/a/com.huawei.solar/test/

That will not work!
"The login credentials provided are not valid."
[image]

reyntjensw, 2023-05-21:
@nissenhj is the server you are using correct?
When you login on the webinterface, does the server correspond?

dsellig, 2023-05-21:
I think he’s using the API credentials whereas he should be using the fusionsolar credentials.

nissenhj, 2023-05-24:
> I think he’s using the API credentials whereas he should be using the fusionsolar credentials.

No, I don't
Today with the new test app it works!

nissenhj, 2023-05-24:
> @nissenhj is the server you are using correct? When you login on the webinterface, does the server correspond?

With this new test app 2.0.6 you cannot see the temperature of the inverter.  Is it something you will put back?  My server is region02eu5.  API login works intermittently while Web login seems to work constantly

reyntjensw, 2023-05-25:
@nissenhj If you want to have the temperature back, you can create a new issue for this.
I'll check if this information is available.

closed 2023-05-28 by reyntjensw