reyntjensw/Homey-FusionSolar #62: Cannot read properties of null (reading 'length')

rlucian, 2023-05-22:
I had one device working perfectly (including battery reading) but since a few versions ago my Homey Huawei device shows as unavailable.  
Tried adding a new device using before 2023 and web credentials. Login seems to go ok but on the "select the devices you would like to add" page i get the error in the issue title.  
Diagnostic report I'd is.  
d56f6236-104b-45a6-8882-b38829f9f7ba

reyntjensw, 2023-05-22:
@rlucian please try out the web version with the web credentials.
If the web credentials do not work, make sure the region is correct

rlucian, 2023-05-23:
Seems to be working, I think the problem was the region.  
However I still have an issue. I have access to 2 plants from the same installer, however, I only got one inverter showing up, and it's not the one where I have the homey setup.   
Let me know if you need additional info.

rlucian, 2023-05-27:
I tried adding another device in homey, but I get "no new devices have been found". 
Any clue if it's huawei or something you could fix?
many thanks!

reyntjensw, 2023-05-28:
@rlucian could you share your credentials via DM (via the Homey community please?)
This should already be ok, but I have no use case for testing

olabil, 2023-05-31:
Hi, 
New here and trying to add a system, SUN2000-8KTL-M1 ,from before 2023. I get the same message when "Devices found":
[image]

I did try to add the web version and after changing the region to "region02eu5" it worked. But I still get the same message trying to add "Systems before 2023". Is there anything I can do?

reyntjensw, 2023-05-31:
@olabil please use the web version of the app. The api will be removed in a couple of weeks.

olabil, 2023-05-31:
> @olabil please use the web version of the app. The api will be removed in a couple of weeks.

Ok got it, will do!

Melmixx, 2023-06-16:
Version 2.0.11. Adding web version (now the only version) and I select region3 as shown in the URL of the website login. However I also just get presented with "Cannot read properties of null (reading 'length'). Tried rebooting Homey, restarting app, reinstalling app.

rlucian, 2023-06-19:
For me nothing worked until I removed the other plant. 
I still get some errors occasionally, but at least it's working again and my main setup works now.
If useful, i can post the errors and generate a debug when it happens again.

jdalesjo, 2023-06-27:
I have the same exact issue as rlucian. Access to two sites but only one shows up, and it is a weird mix of both. The total power is for both sites. One site has battery storage but when I enable battery I get an error and the site becomes unreachable… saying something about “can’t read SOC”

rlucian, 2023-06-28:
I stayed with one site and it works better now. 
I still get an error every now and then, and the device is broken in homey. It seems to happen when the Huawei app is updated but not just then. Restarting the app didn't help, restarting Homey brought the measurements back. Hope this helps you @jdalesjo  

Actually, just checked now and it's broken. Error is "Device unavailable   **Error retrieving data (TypeError: response.data.data is not iterable)**". I created diagnostic report in homey: 29f55e07-9f3c-49da-a1b2-bae9bf7fb0a8  
Tried restarting the app, device with same error, diagnostic report: 97c63629-c2c5-4f7b-a652-8f43ea7cbd46
Restarted Homey altogether, device still broken, so I guess you can ignore my suggestion above. 
App no longer working at all, I see there were some updates yesterday, I guess one of them might have messed something. @reyntjensw let me know if there is any other info i can provide or anything else i should try.

rlucian, 2023-06-28:
I'll try reinstalling the app, as suggested on #66 and see if that helps.

jdalesjo, 2023-06-28:
> I stayed with one site and it works better now. I still get an error every now and then, and the device is broken in homey. It seems to happen when the Huawei app is updated but not just then. [...]

@rlucian How did you select just one site? I can see in the url when using the web interface that the identified system is the top node (my installers ID) and below there are my two sites. NE=34148903 My two sites has ID NE=34963137 and NE=35421964 but those two does not show up when I authenticate the app in Homey, only the top node is visible. However, below the top node all equipment is visible from both sites. 

Is it possible to manually select one of the other nodes? Is there some code I can change somewhere in Homey?