reyntjensw/Homey-FusionSolar #63: App nicht mehr nutzbar

willi-wolke, 2023-05-23:
App läßt sich nicht mehr nutzen!
Version Befor 2023
Fehler:
    Invalid Capability: inverter_temperature, homey:app:com.huawei.solar

[image]

reyntjensw, 2023-05-23:
Can you try this version please? https://homey.app/a/com.huawei.solar/test/

willi-wolke, 2023-05-25:
Getestet mit Version 2.0.7
Neuer Fehler.
Ammeldung funktioniert.
Kapazität wird korrekt angezeigt
[image]
[image]
[image]

reyntjensw, 2023-05-26:
Could you please try the web version, this api version will not be supported anymore.
You need to use the web credentials instead of the api credentials.

willi-wolke, 2023-05-27:
Läuft mit Huawei WEB!

closed 2023-05-28 by reyntjensw