reyntjensw/Homey-FusionSolar #66: Error retrieving data - v2.0.8

JorgenSeemann, 2023-05-29:
Hello Wouter 
After system upgradet to 2.0.8 the web connection throws a new error.
Error retrieving data (typeerror: response.data.data is not iterable).
There is no connection.
Diagnostic: c21b7bcd-24fb-4495-a3aa-af8148331a3b
Restart of app/reboot homey doesnt help
Br
Jørgen

redi78, 2023-06-01:
Hi, i using the Huawei Solar Web method i get an error:
`Error retrieving data (TypeError: Cannot read properties of undefined (reading 'status'))`

JorgenSeemann, 2023-06-01:
Removing the app (2.0.8 test), reboot and reinstall solved my issue

reyntjensw, 2023-06-01:
@redi78 can you try the solution of @JorgenSeemann ?
If this does not solve your issue, could you create a new issue please?

closed 2023-06-01 by reyntjensw