reyntjensw/Homey-FusionSolar #71: connection via modbus

redi78, 2023-06-09:
Hi together,

is there a plan to conect the homey app via modbus instead of fusion solar (api/web)?

reyntjensw, 2023-06-09:
@redi78 no there is not, the current platform for modbus has many different flavours based on the setup

jdalesjo, 2023-06-27:
That’s unfortunate. I have modbus running in homeassistant without any issues. (Custom integration) However, wanted to move to homey and saw the fusion solar integration and had great hopes …

reyntjensw, 2023-06-28:
@jdalesjo I understand your question but what people do not know is how much time these kind of integrations take, let me take you on a journey.

The first api integration was fully documented and it took about 1 to 2 weeks working every evening to get this working.
Huawei changed the api policy and then all the issues started, people had issues with connecting, at a moment it stopped working, people got ratelimited, ...

After this Huawei implemented a second undocumented api endpoint, I spend hours and hours to troubleshoot but without a lot of results. A lot of support request came in to get something working.

That is why I tried the web version because it was the most universal way of getting people to connect, no api user, no more extra issues. (I hoped) This integration is undocumented, so it took me 4 to 5 weeks to convert everything from Python to Nodejs (language that Homey knows)

Adding another way of connecting, supporting users with Huawei (specific) related issues is something that I cannot do.
I cannot spend every evening troubleshooting investigating issues.

This is an open source project and everyone is free to contribute. If someone wants to give it a try, please do.
It's just so much more than using the modbus integration from HomeAssistant, it has to be fully rewritten.

closed 2023-06-28 by reyntjensw