reyntjensw/Homey-FusionSolar #73: Data not up to date

Durchleucht, 2023-06-17:
Hi, I connected my Huawei to Homey via Web. The data forImport from Grid and export to Grid are hours old. This does not help if I want to charge the car. Is there a workaround to get the data uo to date? PV yield is fine for example.
thanks for your help.
Wolfram

jseverinson, 2023-06-19:
I'm also struggling to get updated data. In my case the "Daily yield" just stops updating for many hours, as can be seen in the screenshot below from Homey stats. There are also odd drops every night in the "Monthly yield" data, which recover a few hours later.
[image]

Melmixx, 2023-06-27:
For me the absolute most important value is the "Effekt", which currently says "5 days old". I need to know how much power my solar panels are currently generating, so I can evaluate if my car should charge and by how many amps. Other stuff like Active Usage is almost never older than 5 minutes. I really hope this is sorted soon. Currently using version 2.0.14.

EV-BD, 2023-06-30:
Since (apparently) Huawei's server upgrade last week, I get no data at all. They seem to have changed the web login to require a code confirmation.
The app doesn't need this, do does the Homey app need to authorise like the app rather than the web login? 🤔

reyntjensw, 2023-08-08:
@EV-BD after an upgrade, it is sometimes needed to restart the app, the code confirmation is not handled in the app.
The verification is just a safety check.

jseverinson, 2023-08-08:
I have reinstalled the app and am trying to re-add the device to get it working again, but the pairing interface just says my credentials are invalid. They are not though. Is this because the Huawei servers expect to have that additional captcha verification code?

EV-BD, 2023-08-08:
> @EV-BD after an upgrade, it is sometimes needed to restart the app, the code confirmation is not handled in the app. The verification is just a safety check.

---

Huawei seem to have removed he extra check a couple of days ago and it's now working again. 

So something seems to need the check code to connect your Homey app versus Huawei's own app.

reyntjensw, 2023-08-08:
> I have reinstalled the app and am trying to re-add the device to get it working again, but the pairing interface just says my credentials are invalid. They are not though. Is this because the Huawei servers expect to have that additional captcha verification code?
> 
> Edit: In fact, after restarting the app I get another step but then I get an error message.
> 
> [image]
> 
> Every subsequent attempt I only get an error message saying password is incorrect.
> 
> [image]

Do you get a captcha when you login into the webinterface?

jseverinson, 2023-08-08:
> Do you get a captcha when you login into the webinterface?

Yup, still do.

[image]

reyntjensw, 2023-08-08:
Until the captcha is gone, the integration will not work

jseverinson, 2023-08-08:
Ok that's a bummer. I know there are other Homey apps that show the captcha challenge with an answer field within the pairing session. Would that not work in this case?

reyntjensw, 2023-08-08:
@jseverinson as stated on the Homey community, I'm working on making the app more robust, I'm even looking at the Modbus integration, but that is for later.

Adding the captcha is very hard to do because in the app, I'm using undocumented HTTPS calls to get a login and to get data. So having this is not that easy I'm afraid.

jseverinson, 2023-08-08:
Ok, thanks for all the hard work!