reyntjensw/Homey-FusionSolar #76: Captcha problem

jdalesjo, 2023-07-04:
Can’t login. Noticed captcha requirement on web page, and not supported by integration. No workaround found.

Rippledeedoodle, 2023-07-08:
confirm, login fails due to new captcha in fusion solar login...
Error retrieving data (TypeError: response.data.data is not iterable)

reyntjensw, 2023-07-08:
Since I'm using the calls of the FusionSolarPy library, I can't do a lot with it.
See this existing ticket as well : https://github.com/jgriss/FusionSolarPy/issues/5

Rippledeedoodle, 2023-07-08:
> See this existing ticket as well : jgriss/FusionSolarPy#5

Could it be based on the fact that the App tried to Login too often during
Server maintenance (i saw they activate the captcha during maintenance Last
Time) and should be fine If logged in correctly?

reyntjensw, 2023-07-08:
@Rippledeedoodle that could be but it is a guess, there is no documentation available about the calls.

Rippledeedoodle, 2023-07-08:
@reyntjensw the issue persists, the capcha is queried with every login, no matter what.
if this has to do with suspected misuse, what would be the key to get around?
the only misuse i can imagine would be many logins from my the IP range - which shouldn't be the case.

maybe i will ahve to start looking for a modbus solution and fetch the data inside the network, but iw as really happy to find something to avoid that -.-

reyntjensw, 2023-07-09:
@Rippledeedoodle the only thing you can then do is to remove the app, wait a couple of hours and add it back.

The modbus integration will solve a lot of things but I think this will cause other issues as well (people that will have modbus, people who don't have it). If you have a working version, I'm happy to add it into the app

closed 2023-08-08 by reyntjensw