reyntjensw/Homey-FusionSolar #85: Huawei Region

fredclappen, 2023-08-02:
Hi.

Would it be possible to add the Huawei International Region server?
We are registered on there for South Africa.
When I use one of the EU servers the iSitePower-M unit is picked up but no data populates.

The server is: intl.fusionsolar.huawei.com

Regards
Fred

reyntjensw, 2023-08-08:
Hi Fred

For what integration would you like this?
The luna_web, luna_new or luna_old?

fredclappen, 2023-08-08:
Hi Wouter.

It should be for the Systems after 2023.
What I don’t know is if it would work for the Huawei iSitePower-M unit.

reyntjensw, 2023-08-08:
@fredclappen I don't know if this works with the iSitePower-M unit.
Can you try and install the test version please?

https://homey.app/a/com.huawei.solar/test/

fredclappen, 2023-08-08:
I just tried but get the following error:

An unknown error has occurred [invalid_json]

fredclappen, 2023-08-08:
@reyntjensw 
App installed and it picks up the plant but no data.
I do however not see the option for the intl server in options for either the Systems after 2023 or Web.

I think it might be a change on Huawei’s side that the data is not coming through. I found a project for Home Assistant and that also does not get data but does see all devices of the iSitePower-M. I do know that the iSitePower-M is a new product.

Thank you for the help though.

reyntjensw, 2023-08-08:
I think it is related to the iSitePower-M which might require other calls, but I'm not sure.
Since the HA integration is not showing data either, I'll close this one

closed 2023-08-08 by reyntjensw