reyntjensw/Homey-FusionSolar #86: Cannot login to server for adding Huawei inverter

Bartonstilts, 2023-08-05:
Dag Wouter,

Kan.je me helpen om in te loggen in de Huawei Solar app voor Homey Pro early 2023?

Op de website zelf kan ik inloggen  ( https://eu5.fusionsolar.huawei.com/   )  , maar ik kan niet inloggen via de app in  Homey pro.

Heb je een idee waar dit aan kan liggen?

Alvast bedankt,
Bart

reyntjensw, 2023-08-08:
Hi Bart

Heb je bij de integratie gekozen voor de web variant?

Wouter

Bartonstilts, 2023-08-09:
Hi Wouter,
Ik heb de web variant geprobeerd maar dat lukt niet, daarna heb ik ook de versie voor en na 2023 geprobeerd maar telkens kreeg ik de foutmelding in bijlage.

Heb je enig idee waaraan dit kan liggen?

Met vriendelijke groeten, Best Regards, Cordialement,
Bart Michiels

reyntjensw, 2023-08-09:
Dag Bart

De bijlage komt er niet door, kan je deze via Github posten aub?

Bartonstilts, 2023-08-09:
Dag Wouter,

Ik heb de bijlage via Github gestuurd.
Heb je deze goed ontvangen?

Met vriendelijke groeten, Best Regards, Cordialement,
Bart Michiels

Bartonstilts, 2023-08-31:
Hi Wouter, 

Heb je een idee waarom ik deze foutmelding krijg en hoe ze op te lossen is?

Mvg,
Bart

Dirkvant, 2024-03-17:
Hoi Wouter ik heb hetzelfde probleem net een homey pro aangeschaft en kan geen verbinding maken met Huawei en krijg de zelfde foutmelding heb jij al een oplossing hiervoor of gekregen?

Alvast bedankt
Dirk