reyntjensw/Homey-FusionSolar #87: Huawei Solar is not compatible with the selected Homey

andlisow, 2023-08-06:
Hello,
when trying to install an app using the page https://homey.app/en-gb/app/com.huawei.solar/Huawei-Solar/ , I get the error "Huawei Solar is not compatible with the selected Homey.".
Do I need physical homey device?
Homey app version: 7.2.3.1246, premium
Phone: S21
Android 13.
Thank you,
Andrzej

reyntjensw, 2023-08-08:
Hi @andlisow there is no requirement to have a local device because the app connects to the cloud, but in order to publish for Homey (not Homey Pro), you need to be a company and pay a kind of license https://homey.app/en-us/store/product/homey_developer_athom_12m/.

Since I'm not a verified developer, I cannot publish to Homey.

closed 2023-08-08 by reyntjensw

andlisow, 2023-08-09:
Clear, thank you!