reyntjensw/Homey-FusionSolar #88: Can't add device in Homey Pro 2023

mediacut, 2023-08-08:
Hello,
It's any solution for homey Pro 2023 to make app work again?
[image]

reyntjensw, 2023-08-08:
@mediacut This looks like an issue with your api credentials.
Since a couple of months my api credentials are not working anymore and they are unable to solve it for some reason.

I'd like to fix it but tbh the api nor the web version are stable and reliable.
I'm looking into the modbus option right now but that is not going as planned, connecting and getting it to work on the inverter seems to be very hard.

reyntjensw, 2023-08-08:
> Thanks for fast answer. Our API?

What do you mean?