reyntjensw/Homey-FusionSolar #90: Huawei app not working

mattias-23, 2023-08-11:
I get always this error, and if i put web version it says wrong user or password even if they arw correct

[image]

sunnanfjord, 2023-08-12:
Hi,

The web login is no longer working after Huawei has added a verification code when signing in.

/Daniel

mattias-23, 2023-08-12:
And how can i connect it? Even with system after 2023 it says null is not
an object

sunnanfjord, 2023-08-13:
I was only able to get web login to work before Huawei decided to add verifiication code to the login window.
I also get "null is not an object" if I choose anything else than web login.
At the moment, I can not connect to the inverter at all with this app.

/Daniel

ulrikviebke, 2023-08-21:
I logged in to https://eu5.fusionsolar.huawei.com/ and found that there is a kiosk mode, which enables me to use a readonly view through a simple url (including a hash for my environment). Maybe the app can be rewritten to use the kiosk mode url instead and scrape the data from that view?

sunnanfjord, 2023-08-21:
Actually, the web login seems to be working again since Huawei removed the verification code from the login page.

/Daniel

MorganHolm70, 2023-09-07:
Dear anyone
I am trying to add the Huawei-unit to my Homey-system, but I get the same issue as Sunnanfjord, Mattias and the rest above. or at times the message the the login credentials are not valid. 

I am able to authenticate at the fusionssolar.huawei-online with my credentials. Does anybody know why it wil not be added as a unit in the Homey app?

Many Thanks!