reyntjensw/Homey-FusionSolar #92: Cannot read property 'name' of undefined

Swamulf, 2023-08-16:
Hi, this is my firat time trying the app. When i try to ad the device using Web login i get the above message. What could be the issue here?

ON8SD, 2023-09-21:
Same problem here:
Homey early 2016, weblogin worked fine but without warning I didn't receive any readings anymore. I decided to reinstall the app and since then I have the same error

Tschurl, 2023-10-01:
[image]
aaef5799-6377-4b99-9b68-e6b17a6975b8

christoph-planfox, 2024-01-04:
+1 for the same problem. Cannot add my new Solar System through Homey Pro (2016). Does not matter if I chose System after / before 2023 oder Web - error is always the same. Would love to see that fixed soon :) Thanks!