reyntjensw/Homey-FusionSolar #94: "Power" never updates. Says last updated 26 days ago when it was last installed and configured.

Melmixx, 2023-09-08:
I've installed the app and use it daily. However some values for the device never update. "Active Usage" updates within seconds and is very usable, however Power does not update at all. I've tried reinstalling the app and setting up the device again, but it's the same result every time. Other values that never update for me are Installed Capacity, PV Yield, Daily Yield and Battery Cumulative Discharge. Below you can see that Power has not updated for 26 days. Active Usage is only 7 seconds since last update.

[image]

Does anyone else have this? Is there a fix? Thank you.

lasertack, 2023-12-25:
Same here. 
And at the evening and night hours there is still production shown. It stucks on the latest production value.

Melmixx, 2023-12-27:
For me it turned out to be my Huawei equipment that needed to be updated. Once my installer updated everything worked again.

lasertack, 2023-12-27:
> For me it turned out to be my Huawei equipment that needed to be updated. Once my installer updated everything worked again.

Thanks for the suggestion! 
my equipment is up to date. 

Best regards 

Alex