reyntjensw/Homey-FusionSolar #95: huawei solar integration no longer working: Error retrieving data

SEVDK, 2023-09-29:
**Hello.
My fusionsolar app, which has been working fine, has suddenly reported the following error:
"huawei solar integration no longer working: Error retrieving data (TypeError: response.data.data is not iterable)."
I have installed the web version with region02eu5.
Uninstalling and reinstalling the Huawei app and the web version did not solve the problem. Since I use the app to control when my electric car needs to be charged with solar power, I hope someone can help?

reyntjensw, 2023-09-29:
@SEVDK have you tried loggin in into the fusionsolar website?
It is possible that you have a captcha there which is causing this kind of issues.

SEVDK, 2023-09-29:
Hi Thank you for your reply.
Yes, I have logged on to my Fusionsolar web.
Everything looks fine on the web page.

Willemss001, 2023-10-24:
Same error here..
Can logon to web and data is shown but error in app.

ZGHmn, 2023-11-02:
Hi, I have the same problem: "Error retrieving data (TypeError: response.data.data is not iterable)". Logging on to the website, I see they have applied captcha which I have not seen before (I logon every month to their website). 
My guess is that the captcha is the cause of the problem. Anyone having an idea on how to overcome this issue?

ZGHmn, 2023-11-02:
> Hi, I have the same problem: "Error retrieving data (TypeError: response.data.data is not iterable)". Logging on to the website, I see they have applied captcha which I have not seen before (I logon [...]

Hi, I have the same problem: "Error retrieving data (TypeError: response.data.data is not iterable)". Logging on to the website, I see they have applied captcha which I have not seen before (I logon every month to their website).
My guess is that the captcha is the cause of the problem. Anyone having an idea on how to overcome this issue?

ZGHmn, 2023-11-02:
@reyntjensw Hi, I have the same problem: "Error retrieving data (TypeError: response.data.data is not iterable)". Logging on to the website, I see they have applied captcha which I have not seen before (I logon every month to their website).
My guess is that the captcha is the cause of the problem. Anyone having an idea on how to overcome this issue?