reyntjensw/Homey-FusionSolar #101: Not able to install my system

Martindawall, 2023-12-21:
Hi,
My solar system have no longer a connection to the Homey system. I have tried to re/install the app and restarted my Homey 2023.
I receive this when I try to install the system

[image]

I can see that it has been discussed in this forum, but I can’t find the solution in the present discussions. 

Please can someone help me out.

Code. 8ea68b31-0e7b-4a3b-8209-2496df158dac

Martindawall, 2023-12-21:
I have now looked inside the Fusion solar setup and notes that my system is on region2. So now log/in via the EU5region2 ant now I see my system, but not receiving any data. 
Is there a fix for that …
[image]

hanneskl, 2024-01-18:
I cannot add my system either.
I can still access my system via https://region03eu5.fusionsolar.huawei.com/