reyntjensw/Homey-FusionSolar #104: Not longer able to login or add solar system from web

170KJ, 2023-12-31:
Huawei did make an update to the solar system yeasterday and since that my link to the web stoped working. ¨
Tried to remove the device, remove the app and reinstall it again but no success. First time I try to add device (web, region 0205) it can't find devices and second time it tells me that i have incorrect password or username. 
It seams that the Fusion solar server have moved. Probably that's the reason for the error.

lasertack, 2024-01-01:
Same here. I noticed that the login adress has changed for the Huawei Web interface. So the app is probably trying to to login through the old address.

170KJ, 2024-01-01:
> Same here. I noticed that the login adress has changed for the Huawei Web interface. So the app is probably trying to to login through the old address.

Ok. Hope that the developer of the app is fixing this issue quite soon then!

rmangin-fr, 2024-01-25:
+1 same symptom. I've contacted Fusion Solar support by email (address shown in the communication preceding each server restart).
Answer from Huawei / Fusion Solar: there's no API since the inverter is accessible through "Modbus TCP". Checked out other apps supporting Modbus and it seems there is none for Huawei.
Will check with the developer if we can add an option for local access to the inverter through this "official & LOCAL channel" :)

dragosj, 2024-01-26:
same issue, failed to login and I also think is because of change from "region02eu5" to "uni002EU5"

reyntjensw, 2024-01-29:
If this change https://github.com/reyntjensw/Homey-FusionSolar/pull/105 is tested, I'll release a new version.
I'm not able to test it because of the lack of api credentials

reyntjensw, 2024-02-06:
The test version has been released https://homey.app/a/com.huawei.solar/test/

szferi91, 2024-02-06:
The issue is still the same, I don't see any changes as the list is same as well
[image]
Latest version installed
[image]

lasertack, 2024-02-06:
Same here. Nothing new. Still no chance to add a system.

reyntjensw, 2024-02-06:
Please update to the latest test version

dragosj, 2024-02-06:
tried new test update and same message...nothing added....seems more than just web address change.

lasertack, 2024-02-06:
> Please update to the latest test version

Did delete theropod app from homey and installed the one from you link which is mentioned to be experimental. Same as before

szferi91, 2024-02-06:
V2.0.17 still with the same problem

reyntjensw, 2024-02-06:
This needs further troubleshooting but for now, I don't have time to do this.
If someone else can start troubleshooting, please do so

lasertack, 2024-02-10:
Seems to be an update of the login address. It is now uni00xxx instead of eu... But same error.

Is there something we can do to support you? Monetary or....

reyntjensw, 2024-02-10:
@lasertack the issue is that the api and the web integration are not working for all the users.
The web integration is an unsupported integration by Huawei, only the api is supported.

The api integration might be fixable but Huawei is unable to provide me with a decent api token that works for my inverter (this is the only way I can test)

Help would be grateful from another developer jumping in.

5g11, 2024-02-28:
I received API data from Huawei. What is needed? Is it possible to access the data like HA?

PatrikAnd, 2024-03-09:
I'm not coming in anymore. Getting an error message since late fall and I still haven't managed to connect again. it used to work but now not anymore. I can't find a solution to it.

[image]