reyntjensw/Homey-FusionSolar #110: Why is this app so shitty?

HerrWolf25, 2024-02-21:
This app is so broken. Works only half the time the last 2 weeks it has been completely broken. Why the web shit?
Fix using MODBUS tcp instead. Much better and should be easy to do.

5g11, 2024-02-28:
Can you fix it?

5g11, 2024-03-14:
There will soon be a new app for Huawei.
The first tests are running and look good.

HerrWolf25, 2024-03-14:
Great many thanks you!!!

HerrWolf25, 2024-03-14:
> Here the app. But it's not finished yet. We are testing in the background.
>
>
> https://homey.app/de-de/app/solaredge.modbus/SolarEdge-+-Growatt-TCP-modbus/test/

It WORKS!!!!
Just added the IP.

5g11, 2024-03-14:
The battery is still missing. But when the sun shines there are values.

davidecapra, 2024-03-17:
It doesn't work for me, putting in the inverter's ip doesn't give me any data

5g11, 2024-03-17:
Is the inverter set for TCP Modbus?
Otherwise the data cannot be read.

hansvanveer, 2024-03-17:
> Here the app. But it's not finished yet. We are testing in the background.
> 
> https://homey.app/de-de/app/solaredge.modbus/SolarEdge-+-Growatt-TCP-modbus/test/

Thanks 5g11!
I can confirm I could already connect to my Huawei dongle with the Homeyapp over ModBUS TCP via the link you provided, on port 502 and InverterID 1, in a config of 2 monophase inverters configured to both connect to the Dongle. It took me some checks to find out the IP address of the donle in my network, but after some random tests of unattributed IP addresses, I managed to get reutrn info back.
However, it seems the app is only reading out the values of only one of the two inverters, more specifically the one where the Dongle is plugged in. Changing the InverterdID (0, 2 3 or 4) did not bring any more info.
I hope to be able to provide you some usefull feedback this way. 
Geep up doing the good work!

5g11, 2024-03-17:
Battery is set up. 
It's best to ask further questions in the thread about the app.

HerrWolf25, 2024-04-14:
Hello
App working good!
would it be possible to add possibility to change parameter (power limit )
(zero export limitation)
its to use when the power prices are lower then 0, and you pay money to
export the excess power....

5g11, 2024-04-14:
That is in the works. This version is only only read at first. 
Rather for the German market because selling is a minus business and the focus is on self -consumption.

6 cents feed and buy 26 cents

HerrWolf25, 2024-04-14:
Oh
In Denmark you get money for selling but only when the price is above 0.
Sounds great. Cant wait!

HerrWolf25, 2024-05-23:
Hello
When is the function to change export to zero available?